greiman/SdFat #102: Make the library smaller

masoudr, 2018-04-01:
Hi,
Currently, I'm using this library to just write a simple string to sd card. It works fine but in my project, I don't have enough Flash space. Without the library, it uses about 76.7% of Flash and 59.4% of SRAM in atmega328.
But If I use a simple function to write to sd like this:
```
void sd_log(const char* input){
    //log to sd card
    myFile = SD.open("output.log", FILE_WRITE);
    if (myFile) {
        myFile.println(input);
        // close the file:
        myFile.close();
    }
}
```
It uses 103.8% and 92.3% for Flash and SRAM respectivelly.
Is there any way to reduce the size of Flash so I can fit it into my Arduino pro mini?
Actually, I'm not sure if removing Bootloader may help or not.
I also tried to use your fat16 library but it didn't work with my current hardware.
Thanks.

greiman, 2018-04-01:
Fat16 is optimized for 328 AVR.  What board are you using?

Over the years optimizing SdFat for size has been a top priority.  I don't see a way to save much more memory and support FAT32.

A 512 byte buffer is required to update SD file structures like directories since an entire 512 byte SD block must be written.  Fat16 also has this limit.

You can slightly reduce the flash size by setting USE_LONG_FILE_NAMES zero in SdFatConfig.h.

Here are my size test programs on Uno.

The standard SD.h library based on a very old version of SdFat:
```
/*
 * Program to compare size of Arduino SD library with SdFat.
 * See SdFatSize.ino for SdFat program.
 */
#include <SPI.h>
#include <SD.h>

File file;
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(9600);
  
  // Wait for USB Serial 
  while (!Serial) {
    yield();
  }

  if (!SD.begin()) {
    Serial.println("begin failed");
    return;
  }
  file = SD.open("TEST_SD.TXT", FILE_WRITE);

  file.println("Hello");

  file.close();
  Serial.println("Done");
}
//------------------------------------------------------------------------------
void loop() {}
```
> Sketch uses 10726 bytes (34%) of program storage space. Maximum is 30720 bytes.
> Global variables use 922 bytes (45%) of dynamic memory, leaving 1126 bytes for local variables. Maximum is 2048 bytes.

SdFat with long file names disabled:
```
/*
 * Program to compare size of SdFat with Arduino SD library.
 * See SD_Size.ino for Arduino SD program.
 *
 */
#include <SPI.h>
#include "SdFat.h"

SdFat sd;

SdFile file;
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(9600);
  
  // Wait for USB Serial 
  while (!Serial) {
    SysCall::yield();
  }
  
  if (!sd.begin()) {
    Serial.println("begin failed");
    return;
  }
  file.open("SizeTest.txt", O_RDWR | O_CREAT | O_AT_END);

  file.println("Hello");

  file.close();
  Serial.println("Done");
}
//------------------------------------------------------------------------------
void loop() {}
```
> Sketch uses 8264 bytes (25%) of program storage space. Maximum is 32256 bytes.
> Global variables use 869 bytes (42%) of dynamic memory, leaving 1179 bytes for local variables. Maximum is 2048 bytes.

Note that I have saved over 2KB of flash and about 50 bytes of RAM.

Here is Fat16:
```
/*
 * Program to compare size of Fat16 with Arduino SD library.
 */
#include <Fat16.h>

const uint8_t CHIP_SELECT = SS;  // SD chip select pin.
SdCard card;
Fat16 file;

void setup(void) {
  Serial.begin(9600);
  
  // Wait for USB Serial 
  while (!Serial) {
    yield();
  }
  // initialize the SD card
  if (!card.begin(CHIP_SELECT) || !Fat16::init(&card)) {
    Serial.println("begin failed");
    return;
  }
  file.open("TEST_SD.TXT", O_CREAT | O_EXCL | O_WRITE);
  
  file.println("Hello");

  file.close();
  Serial.println("Done");  
}
void loop(void) {}
```

> Sketch uses 5692 bytes (17%) of program storage space. Maximum is 32256 bytes.
> Global variables use 804 bytes (39%) of dynamic memory, leaving 1244 bytes for local variables. Maximum is 2048 bytes.

Saves lots of flash but only about 60 more bytes of RAM.

greiman, 2018-04-01:
I don't think there is any hope.  The SD libraries need more than 200 extra bytes of stack so I think all of the libraries will crash.

masoudr, 2018-04-01:
@greiman Thanks I used that and it became 99.9% of Flash and 92.3% of SRAM. I think I can reduce my code a little bit. About Fat16, I just got a 4 GB memory card and tried to manually partition it to a lower size below 2GB and then I formatted it to fat16. But on 'if (!card.begin(CHIP_SELECT) || !Fat16::init(&card)) {' it exit. I cant remember right but the value was something negative like (-4 or something). I was not sure if it was related to my SD card or not but the same hardware worked fine on the Fat32 library. Soon I will test your example about fat16 library and report here.
For the last thing, I just need to write the log of my project into an SD card, it doesn't matter if it is formatted to FAT32 or Fat16.

greiman, 2018-04-01:
Once again, you may have crashes since you need 200-300 bytes of free RAM above the amount the IDE prints when you compile.  This is stack memory for temp storage and function calls.

The 4GB card fails because it is SDHC.  The Fat16 library only can access "standard" SD cards.  The SDHC I/O protocol is not supported.

> A SD card will normally work in a SDHC device (possibly with lower performance). SD cards are in the following capacities  4,8,16,32,64,128,256 512MB and 1GB, 2GB only. Currently mainly 2GB SD cards are being sold.
> 
> A SDHC card will not work in a SD device such as a camera or reader. SDHC cards are in the following capacities 4GB, 8GB, 16GB and 32GB

You need to find a 2GB "standard" SD.  It will cost about the same as an 8GB SDHC card.

masoudr, 2018-04-05:
@greiman Finally I got a 2GB memory, I tested it with Fat16 library and now it is working fine with my current hardware and my current utilization is 89.5% of Flash and 88.7% of SRAM. I think it is fine and It uses fewer resources compared to FAT32. I think if someone gets a similar problem to mine, he should use the Fat16 library. Thank you very much.

closed 2018-04-05 by masoudr