greiman/SdFat #105: How to use custom SPI pinout in the Arduino using LowLatencyLogger.ino

williamesp2015, 2018-06-09:
I enabled ENABLE_SOFTWARE_SPI_CLASS in SdFatConfig.h to use Software SPI (like SoftwareSpi.ino) in the 
LowLatencyLogger.ino example nut I got a lot of errors. Anyone knows how to configure custom SPI pins in the LowLatencyLogger.ino example using hardware SPI (like (SPI.begin(CLK_PIN,SO_PIN,SI_PIN,m_csPin)). Thank you

greiman, 2018-06-10:
Software SPI won't work well with the LowLatencyLogger since Software SPI is very slow.

> I got a lot of errors

What are the errors?

williamesp2015, 2018-06-10:
Thanks. I use ESP32 and Arduino platform. I successsflly implemented SdFat using classic hardware SPI port. In another board, I must change the GPIOs so I added SPI begin with my GPIOs but SD.begin failed.
  /*MY MODIFICATION */
    pinMode(SS, OUTPUT);
    digitalWrite(SS, HIGH);
    pinMode(SCK, OUTPUT);
    pinMode(MOSI, OUTPUT);
    pinMode(MISO, INPUT);
    //SPI.begin();
    Serial.println("SPI Begin");
    SPI.begin(SCK,MISO,MOSI,SS);//25,26,27,33
    Serial.println("SPI Begin Transaction");
    SPI.beginTransaction(SPISettings(1000000, MSBFIRST, SPI_MODE0)); // with breadboard, speeds starts with 1MHz 
    /******************************************/
  delay(50); //wait
  Serial.println("SD Begin");
    // Initialize at the highest speed supported by the board that is
    // not over 50 MHz. Try a lower speed if SPI errors occur.
    if (!sd.begin(SS, SD_SCK_MHZ(10))) {
      sd.initErrorPrint(&Serial);
      FileOpened=false;
      fatalBlink();
    }

greiman, 2018-06-10:
You are on your own. I don't have time to help with mods.

williamesp2015, 2018-06-11:
Hi everyone. If your are using Arduino-ESP32 follow this example to use SdFat:
SPI.begin(14,2,15,13);//SCK,MISO,MOSI,ss  change in SdSpiESP8266
Change 50 to 10 in:
configure pins:
  pinMode(SS, OUTPUT);
    digitalWrite(SS, HIGH);
    pinMode(SCK, OUTPUT);
    pinMode(MOSI, OUTPUT);
    pinMode(MISO, INPUT);
if (!sd.begin(SS, SD_SCK_MHZ(10))) {
Remove all these:
    //  while (!Serial.available()) {
    //    SysCall::yield();
    //  }
Enjoy using ESP32 by custom or standard Pinout

greiman, 2018-06-11:
Glad to see you found a fix.  I don't have a working ESP32 so I would have never been able to guess the speed problem.

I am puzzled why these line are necessary.  SdFat sets the mode and level of chip select and I would expect the ESP32 SPI library begin() function to initialize SCK, MOSI, and MISO.

```
pinMode(SS, OUTPUT);
digitalWrite(SS, HIGH);
pinMode(SCK, OUTPUT);
pinMode(MOSI, OUTPUT);
pinMode(MISO, INPUT);
```

ModistESP32, 2019-05-15:
Hey there,
I'm trying to make this work on my ESP32 TTGOV2 board. It's using the same SD pins as williamesp2015 postet so SD.begin(14,2,15,13). But I dont get your changes that i have to do in the "SdSpiESP8266" file. I assume it is quite simple but I am not sure what to do. I will go for try and error but I hope someone can explain. Best regards.

fotosettore, 2019-10-02:
Please may you explain better how to use esp32 compatibility about sdfat overall using it with 2 sd card and possibility to change miso mosi cs and sck ?
Just a minimal example.
Many thanks

2shaar2059, 2020-08-06:
For Teensy4.0, Im getting other issues like createContiguous file errors but I solved the following error:

FreeStack: -1618084
Records/block: 42
Can't access SD card. Do not reformat.
SD errorCode: 0X30,0X2 

by editing the clock speed:

if (!sd.begin(SS, SD_SCK_MHZ(26.66666599999))) 

If I go any high than 26.66666599999 , I get the "Can't access SD card. Do not reformat." error again.

greiman, 2020-08-06:
Use [SdFat-beta](https://github.com/greiman/SdFat-beta).

gustavoser78, 2021-06-14:
> Hi everyone. If your are using Arduino-ESP32 follow this example to use SdFat:
> SPI.begin(14,2,15,13);//SCK,MISO,MOSI,ss change in SdSpiESP8266
> Change 50 to 10 in:
> configure pins:
> pinMode(SS, [...]

hi!  I´m using ESP CAM kit. I do need to format the SD Card. So I look for and find SdFat is the only library with this funcionality.  I tried what you suggest above but it is not working.  Should it work in SdFormatter too?
Thanks