greiman/SdFat #106: TeensySdioDemo working only in EX mode

ecurtz, 2018-06-21:
I'm not sure if this is actually a bug, or me being ignorant of the differences in the Ex versions of the classes, but when I run the TeensySdioDemo on a Teensy3.6 it works fine in SdFatSdioEX mode, but locks up (no error, just never finishes the first write) in SdFatSdio mode. It doesn't seem to matter which I run first. Kingston 2 GB card, I'll try another card as soon as I can manage to find one.

```
SdFatSdioEX uses extended multi-block transfers without DMA.
SdFatSdio uses a traditional DMA SDIO implementation.
Note the difference is speed and busy yield time.

Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,6759.59,13927.58
1024,5889.88,13994.28
2048,5923.13,13974.54
4096,5387.51,13867.97
8192,5918.10,13867.44
16384,5934.83,13843.57
32768,5913.20,13828.42

totalMicros  14112449
yieldMicros  542264
yieldCalls   108
yieldMaxUsec 139346
kHzSdClk     45000
Done
Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,
```

ecurtz, 2018-06-21:
Same result with a 8GB SanDisk card.

greiman, 2018-06-22:
I tried a 16 GB Sandisk Extreme microSD with Teensy 1.42 applied to Arduino 1.8.5. 

Here are the results.

> SdFatSdioEX uses extended multi-block transfers without DMA.
> SdFatSdio uses a traditional DMA SDIO implementation.
> Note the difference is speed and busy yield time.
> 
> Type '1' for SdFatSdioEX or '2' for SdFatSdio
> 
> size,write,read
> bytes,KB/sec,KB/sec
> 512,17683.42,18738.21
> 1024,17621.42,18864.20
> 2048,18193.04,19079.72
> 4096,18030.83,19131.37
> 8192,18063.87,19193.31
> 16384,17881.85,19264.01
> 32768,18338.00,19276.27
> 
> totalMicros  6345877
> yieldMicros  182016
> yieldCalls   183
> yieldMaxUsec 3890
> kHzSdClk     45000
> Done
> Type '1' for SdFatSdioEX or '2' for SdFatSdio
> 
> size,write,read
> bytes,KB/sec,KB/sec
> 512,462.30,2024.71
> 1024,1113.76,2584.53
> 2048,2618.79,4718.25
> 4096,4884.26,8747.82
> 8192,7053.63,12601.54
> 16384,8041.72,15907.25
> 32768,11628.78,18295.37
> 
> totalMicros  45328798
> yieldMicros  44908536
> yieldCalls   66942
> yieldMaxUsec 76172
> kHzSdClk     45000
> Done
> Type '1' for SdFatSdioEX or '2' for SdFatSdio
> 

SDIO DMA  is very slow compared to SDIO EX mode.  It took about 15 seconds for the write with a 512 byte buffer to finish.  a slower card can take 60 seconds.

I tried an old Kingston 2 GB microSD and it took almost a minute to finish the 512 byte write.

> size,write,read
> bytes,KB/sec,KB/sec
> 512,149.20,1906.54
> 1024,312.69,2844.51
> 2048,701.25,4684.77
> 4096,3024.24,6749.39
> 8192,5326.35,9063.67
> 16384,6241.27,10111.03
> 32768,7875.14,10942.01
> 
> totalMicros  114676684
> yieldMicros  114160728
> yieldCalls   80388
> yieldMaxUsec 124264
> kHzSdClk     45000
> Done

ecurtz, 2018-06-22:
You were right, it's just terrible write performance in non EX mode - but this card is an order of magnitude worse than the one you tested at write sizes below 512 bytes - I reduced the file size to 512k and got these results. Sorry for the false report.

```
Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,8.53,1283.88
1024,10.03,2274.00
2048,749.87,3876.55
4096,840.70,5992.75
8192,637.10,8072.55
16384,1069.43,9408.49
32768,1342.30,10895.88

totalMicros  117776635
yieldMicros  119161350
yieldCalls   4275
yieldMaxUsec 399823
kHzSdClk     45000
Done
```

closed 2018-06-22 by ecurtz

ecurtz, 2018-06-22:
Just a final follow up - I reformatted the card with the official app from sdcard.org and got numbers much closer to the ones Bill posted (half the write speed with small blocks). Never trust your memory about how old cards were formatted.