greiman/SdFat #107: Reading/writing in subdirectories doesn't work.

dulsi, 2018-06-23:
When I try to read or write a file in a subdirectory it doesn't work. Reading or writing in the root directory works. A different version of the SdFat library (from SuperOteme not in this repository), it works. I believe this is due to SdFile.cpp. At line 782 and 1452, you have:

      if (block <= m_vol->cacheBlockNumber()
          && block < (m_vol->cacheBlockNumber() + nb)) {

The working code has:

      if (m_vol->cacheBlockNumber() <= block
          && block < (m_vol->cacheBlockNumber() + nBlock)) {

I believe this version is correct as you want to check that the block is inside the cached data. When I changed the code, it started working.

greiman, 2018-06-23:
Thank you.  Excellent bug find.

I have updated the repository.  See "Fix cache flush/invalidate bug".

theshadow124, 2018-07-12:
Hello guys, I am still unable to write to subdirectories using the most recent version of the lib in the repository, I have not been able to narrow down a solution as of yet.

theshadow124, 2018-07-13:
I posted here last night, but I had a few more ideas to try, but to no effect. I am unable to write to subdirectories even with the newest version of the lib.

Right now I am trying to write to a sub dir and it hangs the entire micro, but to the root its fine.

greiman, 2018-07-13:
To: theshadow124

> Right now I am trying to write to a sub dir and it hangs the entire micro, but to the root its fine.

There are several reasons possible.  You need to provide more info about your board and a simple example that fails.

A common cause of hangs is limited memory.  Does the DirectoryFunctions example work?

theshadow124, 2018-07-14:
Hello Greiman,

I have finally figured out what my problem is! But alas I do not know how to fix it, basically, I am building a weather logging device that wakes up periodically and saves the current conditions to an SD card.

I have a pin that controls power to the SD card, so whenever I am not talking to the SD card I turn the power off and set the CS low to try to conserve power while in sleep mode. This probably shouldn't be a discussion under this issue.

I finally figured this out a few min ago... and it appears that if I turn the SD card power off at all it will no longer talk when I wake back up. Is there any proper way to re-init the SD card after powering it off, or removing it and putting it back in? If i call begin a second time it hangs the entire micro.

greiman, 2018-07-14:
Before you power off the SD, close any files open for write.  

After power is restored, you must reinitialize the SD by calling begin().

The SD card has a RISC processor that must be rebooted and initialized after power is restored.

[Here ](https://thecavepearlproject.org/2015/11/05/a-diy-arduino-data-logger-build-instructions-part-4-power-optimization/)is a project that uses three AAA batteries to log data in caves for up to a year.

The cave logger tutorial starts [here](https://thecavepearlproject.org/2015/10/24/diy-arduino-logger-build-instructions-part-1/).

theshadow124, 2018-07-16:
Thank you for that insight, I am still getting some weird behaviour. I have made sure I am closing all files before opening other files or turning the SD card off.

I also wrote a subroutine to check if I can create a temporary file before doing any work and if not, close the file and power cycle the card and repeat. But an odd thing that happens, is I open this ".tmp" file, close it and delete it, and then try opening another file and the second operation can still fail... sometimes. Ideas?

theshadow124, 2018-07-16:
Do you want me to create a separate issue for this as I narrow down whats going on?

Another odd thing that happens along with seemingly randomly not being able to save files, I tried to create a folder, and it just created a file instead? I am having trouble coming up with a reliable test file for this though, it seems to just happen randomly. I have tried multiple SD cards formatted as FAT32, and with both 8 char long filenames and without.

greiman, 2018-07-16:
I can't help since I don't have your hardware. 

> I have a pin that controls power to the SD card, so whenever I am not talking to the SD card I turn the power off and set the CS low to try to conserve power while in sleep mode. 

I hope you are not controlling power to the SD with just a pin.  SD cards draw lots of current for a brief time when programming flash.  You need a transistor switch to reliably switch power to an SD.