greiman/SdFat #108: What might cause SD.begin method to halt?

vicatcu, 2018-07-06:
SD.begin is supposed to return true when initializing an SD card correctly, and false if a card is absent or initialization fails, but I have an Arduino which exhibits the odd behavior of halting on a call to begin with no SD card present. Many other Arduino units do not exhibit this behavior, so it must be _some_ kind of hardware defect in the unit that is halting, but I'm not sure what. that could be. From a software standpoint, can you articulate some (set of) condition(s) that could cause SD.begin to _not_ return? I would have expected a timeout to happen in the absence of an SD card followed by a return of false.

greiman, 2018-07-07:
You have not described the Arduino, other software, and connected hardware so I will assume it is a bare Arduino like an Uno.

During initialization, commands are tried multiple times and timed out using millis().  If interrupts are disabled or the timer used by milis() is not running, the timeout will fail.  

Another unlikely possibility is that the SPI controller/library has a problem and hangs.

vicatcu, 2018-07-08:
Thanks @freiman, it's actually a WildFire board (docs etc. @ wildfire.wickeddevice.com), but your insights are appreciated and relevant in any case.

greiman, 2018-07-09:
I can't guess what might be happening with all the stuff on this board.

vicatcu, 2018-07-10:
yea I didn't expect you to, thanks again for sharing your insights on what begin() does.

closed 2018-07-10 by vicatcu