greiman/SdFat #110: Metro M4 SPI / SdFat performance issue

SapientHetero, 2018-08-09:
I just started working with the Metro M4. I mounted an Adafruit WINC1500 shield and ported Arduino-based software I wrote for a Mega to it expecting huge performance improvements, only to find that it crawls. I traced the slowdown to portions of my code that communicate with the shield, so I ran the SdFat bench example program to quantitatively compare the M4's performance to that of the Mega using the same shield, same software and same freshly formatted micro-SD card for both tests. The M4 performance on this test was on the order of 28x slower for writes and 50x slower for reads.

I haven't yet done quantitative benchmarking on WiFi performance but the software I ported to the M4 reports on its maximum loop time. When idling, the max loop time on the Mega is 2ms; on the M4, it's nearly 900ms. I've isolated the slowdown to a function that does WiFi communication, which leads me to believe the issue is SPI-related.

I've attached the version of SdFat bench I used. I modified it to use "SPI_FULL_SPEED" after my first run using the original SPI_HALF_SPEED setting produced poor results, but the results were virtually identical at each setting.

Any insights you might have into this would be deeply appreciated.

bench.zip

MEGA 2560 performance
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 6711
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
281.23,27624,1572,1813
284.88,95152,1540,1790

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
507.68,3108,988,1001
507.84,1992,988,1001

Metro M4 performance
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 192112
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
10.30,344156,49334,49677

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
10.69,95857,47883,47897
10.69,95857,47883,47897

Done

I tried running this test with the same shield, same SanDisk Ultra 32GB SDHC UHS-I micro-SD card and an Arduino DUE and got the following result:

Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 91576
Can't access SD card. Do not reformat.
SD errorCode: 0X50,0X6

According to SdInfo.h, 0x50 is a SD_CARD_ERROR_READ; not sure how to interpret the 0x6.

I also ran SdFormatter on the DUE and it tells me "error: cardSize".

After getting this error I moved the shield back to a Mega and everything still works fine.

I buzzed out the SPI connections on my shield with a meter and none seem to have excessive resistance. Both the DUE and the Metro M4 are 3.3V devices, while the Mega runs at 5V. Could it have something to do with this?

The version of bench used in these tests is attached.
[bench.zip](https://github.com/greiman/SdFat/files/2275850/bench.zip)

greiman, 2018-08-09:
I have no idea what is happening with the Metro M4.  I have never used one.  Something is clearly wrong when 50ms is the average latency for a 512 byte transfer.

On Due I connect the SD directly to the SPI and chip select pins.  No 5V to 3.3V level shifters.

512 byte transfers are slow with new high performance cards like the Ultra since they only emulate 512 byte blocks.  The SdFatEX class is much faster.

Here are some results for the bench example on my Due with a 32GB Ultra.

Due with only mod:
const uint8_t chipSelect = 10;

```
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SL32G
Version: 8.0
Serial number: 0XC2E6F10D
Manufacturing date: 6/2014

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
299.88,87291,1089,1704
308.39,71190,1094,1657

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1493.33,1000,329,341
1493.33,1001,329,341
```
Here is SdFatEX.

change SdFatConfig.h 
#define ENABLE_EXTENDED_TRANSFER_CLASS 1

Comment out first line and remove comment from second line in the bench example.
//SdFat sd;
 SdFatEX sd;

```
File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
4190.85,30248,109,119
4336.24,21754,109,115

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
4616.51,1203,108,109
4620.78,1206,108,109
```
Te average time for a 512 byte transfer is  109 - 119 microseconds.

SapientHetero, 2018-08-12:
Just ran 3 versions of the SD card info example sketch on my DUE with the same shield and micro-SD card mentioned above. The 3 versions are the Arduino SD library, v 1.0.6 of SdFat and v2018.6.24 of SdFs. The Arduino library worked fine, but both SdFat and SdFs failed to initialize the card.  Serial output from all 3 follow. All 3 work on my Mega. Any thoughts about what the problem might be?

**DUE output for Arduino Sd.h CardInfo example sketch**

Type any character to start

Initializing SD card...Wiring is correct and a card is present.

Card type:         SDHC
Clusters:          973584
Blocks x Cluster:  64
Total Blocks:      62309376

Volume type is:    FAT32
Volume size (Kb):  31154688
Volume size (Mb):  30424
Volume size (Gb):  29.71

Files found on the card (name, date and size in bytes): 
BENCH.DAT     1980-00-00 00:00:00 5000000

**DUE output for SdInfo example sketch (SdFS version=2018.6.24)**

Disabling SPI device on pin 10

Assuming the SD chip select pin is: 4
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 11 ms
readInfo failed
SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x17
SD errorData = 0x1

type any character to start

**DUE output for SdInfo example sketch (SdFat v1.0.6)**

error: cardSize failed
SD errorCode: 0X50,0X20

SdFat version: 1.0.6

Disabling SPI device on pin 10

Assuming the SD chip select pin is: 4
Edit SD_CHIP_SELECT to change the SD chip select pin.

type any character to start
error: cardSize failed
SD errorCode: 0X50,0X20

type any character to start

greiman, 2018-08-12:
These errors may be caused by the fact that SdFat and SdFs uses a very fast custom DMA SPI driver.

In bench.zip you used:
```
  if (!sd.begin(chipSelect, SPI_FULL_SPEED)) {
    sd.initErrorHalt();
  }
```
This will result in an SPI speed of 48 MHz on Due since this sets the SPI speed to the max <= 50 MHz.

`#define SPI_FULL_SPEED SD_SCK_MHZ(50)`

The WINC1500 shield has a level converter that may be too slow.

The ten year old version of SdFat in SD.h uses the Due library SPI driver and the default SPI speed for SD.h is 4 MHZ.

Try using the standard Due SPI library and replace the speed argument in begin with 4 MHz like this:

Edit SdFatConfig.h and set USE_STANDARD_SPI_LIBRARY to one.

`#define USE_STANDARD_SPI_LIBRARY 1`

In your program replace SPI_FULL_SPEED with SD_SCK_MHZ(4).
 ```
  if (!sd.begin(chipSelect, SD_SCK_MHZ(4))) {
    sd.initErrorHalt();
  }
```

SapientHetero, 2018-08-12:
Thanks so much for your help. I started at 4 MHz and increased speed until it stopped working above 40 MHz.

My DUE bench sketch results (SdFat 1.0.6)  at 40 MHz with a 65536 byte buffer were:

Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: 26548
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SC32G
Version: 8.0
Serial number: 0X111A0968
Manufacturing date: 2/2012

File size 5 MB
Buffer size 65536 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
621.66,131325,103974,105336
623.84,118452,103765,104967

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
654.93,101129,99958,100060
655.02,101035,99959,100058

Done
Type any character to start

SapientHetero, 2018-08-12:
I notice that the SdFs version of "bench" doesn't use the SD_SCK_MHZ(50) parameter used in the SdFat version. How can I reduce the SPI clock speed in SdFs?

BTW, I got SdFat bench to run on my Metro M4 by reducing the SPI clock speed to 12 MHz. throughput is still faster than either my Mega or Due, but not as fast as your SdFs results on the Due.

closed 2018-08-17 by SapientHetero

greiman, 2018-08-18:
You can use SD_SCK_MHZ(50) with SdFs.  Guess I didn't provide many examples.

Define SD_CONFIG like this where CS_PIN is your chip select pin and SPI_OPTIONS is DEDICATED_SPI or SHARED_SPI.
`
#define SD_CONFIG SdSpiConfig(CS_PIN, SPI_OPTIONS, SD_SCK_MHZ(50))`

Use this call for begin:

`  sd.begin(SD_CONFIG);`