greiman/SdFat #113: ESP32

siara-cc, 2018-10-10:
Hi, great library, especially support for multiple cards.  Do you plan to support ESP32 as well?

siara-cc, 2018-10-10:
I just noticed ESP32 already has support for SD, SD_MMC, SPIFFS and also has implemented a VFS with multiple mount points to access these from standard fopen interfaces.  So I guess implementing SdFat for ESP32 is redundant. Thanks.

closed 2018-10-10 by siara-cc

Sebastiangka, 2019-12-17:
@siara-cc, do you know about a library to write on an SD-Card for the esp32 with good write speed?
I´m developing my application with Arduino IDE.
Thanks!

greiman, 2019-12-17:
The esp32 is fairly slow for SD write because of poor SPI performance.

I tried to improve it with dedicated SPI in [SdFat-beta](https://github.com/greiman/SdFat-beta).  

Here is the bench example at 30 MHz SPI clock, 240 MHz CPU.  It may be unstable at this speed.  It fails at higher SPI speeds.

```
Type any character to start
Type is FAT32
Card size: 32.01 GB (GB = 1E9 bytes)

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Version: 1.0
Serial number: 0X1550FE3
Manufacturing date: 10/2015

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2868.62,746,176,177
2870.26,200,176,177

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2702.70,196,187,188
2705.63,196,187,188
```
This speed depends on using dedicated SPI.  The version 1.0.4 of ESP32 only defines one SPI port so that can be a problem.

For shared SPI the speed drops a great deal.

```
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
454.05,20275,235,1126
447.95,22600,233,1142

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
922.00,1946,417,554
936.50,1902,419,545
```

Teensy 4.0 gets 22 MB/sec read and write.  It has 4-bit wide SDIO at 50 MHz.  About 10X faster than the fastest ESP32.

siara-cc, 2019-12-18:
FYI ESP32 also has a dedicated 4-bit SDIO port.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/sdio

greiman, 2019-12-18:
> FYI ESP32 also has a dedicated 4-bit SDIO port.

This example uses of two ESP32 modules, one a slave that is dedicated to the SD and a second host module that requests operations from the slave.

This is not a true SDIO controller, it emulate the features of sophisticated controllers that other chips have using a second ESP32 board.  Many ESP32 boards won't work because of conflicts with pin usage.

I looked at examples and it is used for raw write/read to an SD with no file system.

I don't see any simple way to use this with Arduino in a file system.

I looked at info I found [here](https://www.esp32.com/viewtopic.php?f=2&t=849&start=20) and it would be very slow if used in a filesystem.  with single sector writes it runs at 0.02 MB/sec.

The above benchmark is for single sector writes.

greiman, 2019-12-18:
I used ESP32 1.0.4, I ran the ESP32 SD_Test example using the same SD and hardware as with the SdFat bench example.

Here is the result for the ESP32 SD.h library.  It uses 512 byte transfers.

```
1048576 bytes read for 2885 ms
1048576 bytes written for 4402 ms
```
read: 363.46 KB/sec
write: 238.20 KB/sec

Sebastiangka, 2019-12-18:
I got the same measure using the standard SdFat-Library, been aware that the SPI-Clock shouldn't be higher than 30MHz. That's why I was looking for a faster solution to write on the SD. Thanks for your quick answer @greiman , I'm still amazed about the improvement in the Beta version. I have been watching it the whole day trying to understand it so I could make good use of it.

Do you have any advice for me to be able to understand better the good use of SPI? 

I intend to use a circular buffer in the ESP32-RAM with 512 or 1024 Block, then transfer the data to the SD. Good for me its that the SD is the only Slave-SPI Device.

greiman, 2019-12-18:
The beta version of SdFat uses buffer in the SD card, 512 KB is the size of an SD record unit.  512 byte sectors are just emulated, 1024 to a flash page in the SD. You can't improve much in your app.

I believe buffering should be handled in the library.  The app should do what ever is a natural size write.

On other issue with SD cards.  there can be an occasional long write latency.  This can't be controlled since it is associated with internal moves of data by the wear leveling algorithm.

You can buffer data and write when the SD is not busy.  See the [ExFatLogger](https://github.com/greiman/SdFat-beta/tree/master/examples/ExFatLogger) example.

I have a second layer of buffering for smaller writes.  Here is the result for 10 byte read/writes.  This is fast because the ESP32 CPU is fast. 1651 KB/sec requires 165,100 10 byte writes per second.  That is an amazing number of I/O calls per second!

```
Type is FAT32
Card size: 32.01 GB (GB = 1E9 bytes)

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Version: 1.0
Serial number: 0X1550FE3
Manufacturing date: 10/2015

FILE_SIZE_MB = 5
BUF_SIZE = 10 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1651.80,197,1,5
1654.53,197,1,5

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1607.72,197,1,5
1609.27,197,1,5
```

bybtech, 2020-02-17:
Hi, 
congrats for your work! @greiman 
I am going totally crazy in order to set up a decent speed for a microSD card on an ESP32-Wroom-32D. SdFat helped me a lot with this! 
I am programming directly the module without any dev board (I used Adafruit Huzzah32 before but the exposed pins are only the VSPI pins)

- First issue: In order to get higher writing performances and can use also the SD card interface with the same wirings I selected the pins of HSPI >> pins I/0: 2(miso), 15(mosi), 14(sck), 13(cs).
Do these wirings correspond to the best possible choice?

- Second issue: How I can set the SD card interface 1bit mode with the SdFat library?
I was able to do that with #include "SD_MMC" library but not with SdFat library.
Which statements and initializations are needed?

Many thanks!
Enry

RodEnry, 2020-02-17:
Hi,
congrats for your work! @greiman
I am going totally crazy in order to set up a decent speed for a microSD card on an ESP32-Wroom-32D. SdFat helped me a lot with this!
I am programming directly the module without any dev board (I used Adafruit Huzzah32 before but the exposed pins are only the VSPI pins)

First issue: In order to get higher writing performances and can use also the SD card interface with the same wirings I selected the pins of HSPI >> pins I/0: 2(miso), 15(mosi), 14(sck), 13(cs).
Do these wirings correspond to the best possible choice?

Second issue: How I can set the SD card interface 1bit mode with the SdFat library?
I was able to do that with #include "SD_MMC" library but not with SdFat library.
Which statements and initializations are needed?

Many thanks!
Enry

greiman, 2020-02-17:
I can't help with ESP32 advice.   I only test with the default SPI pins.

When I want speed I use Teensy 4.0.  It gets about 22 MB/sec read/write with 4-bit SDIO.

Other chips have faster DMA SPI.  An old Arduino Due does about 4.5 MB/sec read/write.

SPI for an SD card is always 1-bit.

ESP32 is low priority now.  I want to release the beta as SdFat V2 for standard Arduino boards then add features for other boards.

RodEnry, 2020-02-17:
> I can't help with ESP32 advice. I only test with the default SPI pins.
>
> When I want speed I use Teensy 4.0. It gets about 22 MB/sec read/write with 4-bit SDIO.
>
> Other chips have faster DMA SPI. An [...]

Thank you for the reply!
Not speaking about the EPS32 now,
how I can use your library to interface an SD card without SPI (in 1 bit mode)?
I can't find the initialization and declaration need. I was able to use only the SDMMC library.
Thank!

greiman, 2020-02-17:
SdFat only supports 1-bit SPI mode.  Support for 1-bit SDIO mode would require a driver.

I have no interest in 1-bit SDIO.

RodEnry, 2020-02-18:
> SdFat only supports 1-bit SPI mode. Support for 1-bit SDIO mode would require a driver.
> 
> I have no interest in 1-bit SDIO.

Ok, thank you! I think I solved the problem, finally!
I am doing this:
file = SD.open(filename, FILE_WRITE);
file.preAllocate(30000000);
----------------------------
SAMPLING at more than 1000Hz - approx 42 bytes
----------------------------
file.truncate();
file.flush();
file.close();

It seems that is going really well! Putting a stupid console timer seems that the time for writing a small block is about 200 us not at every lap.
It seems that the writing is done once 512 bytes are collected. The pattern is the following.
2
2
1
229
2
2
1
2
2
1
1
1
2
228
2
2
2
2
2
2
2
1
1
2
229
2
2
2
2
1
1
2
2
1
1
229
2
.
.
.

greiman, 2020-02-18:
Yes SDs have block write commands that write multiples of 512 bytes.  

SD cards may take over 100 ms to do a write if they are busy moving data for "flash leveling" to spread wear of flash pages.  This can not be controlled externally.

RodEnry, 2020-02-28:
Thank you for the answer @greiman!
How is possible to use two SPI bus at the same time without interfering each other?

It seems to be impossible.
---------------------------------------------CODE-------------------------------------------------
#include <Arduino.h>
#include "SPI.h"
#include "SdFat.h"
#include "Mcp320x.h"

//SD card pin definition HSPI
#define SD_MOSI 15
#define SD_MISO 2
#define SD_SCK 14
const uint8_t SD_CS_PIN = 13;                      
SdFat SD;
File file;
#define SPI_CLOCK SD_SCK_MHZ(25)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)      //SdFat 2 library
bool   SD_present = false;
SPIClass sdSPI(HSPI);

//External ADC pin and init definition VSPI
#define ADC_MOSI 18
#define ADC_MISO 19
#define ADC_SCK 5
const uint8_t ADC_CS_PIN = 4; 
#define ADC_VREF    3300     // 3.3V Vref
#define ADC_CLK     3000000  // SPI clock MHz
#define SWSPL_FREQ  10000    // sample rate 10 KHz
MCP3208 adc(ADC_VREF,ADC_CS_PIN);
SPIClass adcSPI(VSPI);

void setup() {
    Serial.begin(115200);
    pinMode(ADC_CS_PIN, OUTPUT);  
    pinMode(23,OUTPUT);
    digitalWrite(23, HIGH);
   
    //Init SD
    sdSPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS_PIN);
    if (!SD.begin(SD_CONFIG)) {
        Serial.println("Can't Initialize SD CARD ");
        SD_present = false;
    } else {
        Serial.println("SD card ready ");
        SD_present = true; 
    }
    delay(2000);
    file = SD.open("Test",FILE_WRITE);

    //Init MCP3208
    SPISettings settings(ADC_CLK, MSBFIRST, SPI_MODE0);
    adcSPI.begin(ADC_SCK,ADC_MISO,ADC_MOSI,ADC_CS_PIN);
    SPI.beginTransaction(settings);
}
---------------------------------------------CODE-------------------------------------------------

Any idea to make SdFat working in this context?
Cheers!

greiman, 2020-02-28:
The ESP32 chip has four SPI controllers, [see this](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html).  Only two appear to be usable for user programs.

I don't know how the ESP32 Arduino package accesses these ports.

On other boards with multiple SPI controllers there is a way to specify ports.  See the [STM32Test ](https://github.com/greiman/SdFat/tree/master/examples/STM32Test)example.

Other boards can directly use SPI1, SPI2, etc.

I have no plans to extend this version of SdFat since it will soon be replaced by V2, [now in beta](https://github.com/greiman/SdFat-beta).

ha11otronix, 2024-02-22:
I see this is being so old closed thread but I just wanted to see if now we have new ESP boards like ESP32-S2 and ESP32-S3 which has SDIO interface if we can use this library with those chips or not?

greiman, 2024-02-22:
> I see this is being so old closed thread but I just wanted to see if now we have new ESP boards like ESP32-S2 and ESP32-S3 which has SDIO interface if we can use this library with those chips or not?

I looked at the ESP drivers and found I could go faster with dedicated SPI.

The problem with modern cards is the size of flash pages which are some sizable power of two multiple of 16K bytes and the huge internal RAM buffers.

In dedicates SPI mode and the SDIO drivers I have written, I maintain read or write mode as long as possible.  In this way  you can read or write several GB as a single transfer.

The ESP driver does not maintain read or write mode so small, less than a flash page, writes causes a flash page to be read, updated and written to a new flash page.  This is extremely slow. 

Read has a similar problem but less severe.  The card must read an entire flash page which may be 64kB or more for a 512 byte read.

Many MCUs have SDIO controllers that don't allow the above since DMA automatically terminates the transfer.

ha11otronix, 2024-02-22:
I see so the way ESP32 is handeling DMA for SDIO is not making this library having support.

greiman, 2024-02-22:
> I see so the way ESP32 is handeling DMA for SDIO is not making this library having support.

This is a general problem for most MCUs .  It is rare for the SDIO controller to support such transfers with DMA.  I use programed I/O on Teensy boards.  this works since the Teensy controller has a 512 byte FIFO 32-bits wide so I can do a transfer in 5 μs.

I did a SDIO for RP2040 as a pio driver so it does exactly what I need.