greiman/SdFat #115: CreateContiguous and preErase

mjeanrichard, 2018-11-06:
Hi,

I am a bit confused as how the raw write should work. I tried with the following code:

```cpp
#include "SdFat.h"

SdFat _sd;
SdFile _file;

uint32_t bgnBlock = 0;
uint32_t endBlock = 0;

const uint16_t BLOCK_COUNT = 50000;

void loop() {
	_sd.remove("test.bin");
	if (!_file.createContiguous("test.bin", 512UL * BLOCK_COUNT)) {
	  Serial.println("createContiguous failed");
	  while(true){}
	}
	if (!_file.contiguousRange(&bgnBlock, &endBlock)) {
	  Serial.println("contiguousRange failed");
	  while(true){}
	}
	Serial.println("Ok, new file open.");
	if (!_sd.card()->writeStart(bgnBlock, BLOCK_COUNT)){
	  Serial.println("Block Start failed!");
	  while(true){}
	}
	if (!_sd.card()->writeStop()){
	  Serial.println("Block Stop failed!");
	  while(true){}
	}
	_file.close();
	Serial.println("Ok, cleared file...");

	while(true){}
}
```
 
 It sucessfully creates a file (25k), but it is not empty. Did I missunderstnad the second parameter of writeStart? Shouldn't this erase the file?
 
Thanks for your help!
Meini

greiman, 2018-11-06:
The file will be the size you create.  FAT does not allow preallocation.

You then rewrite the file using writeStart(), writeData(), writeStop().  Unfortunately you are limited to writing 512 byte binary blocks.  

I no longer use this method in my apps.  I am developing a new version of SdFat that supports exFAT.  exFAT allows preallocation.  

[Here ](https://github.com/greiman/SdFs)is an early version of the library I am modifying to be more compatible with SdFat. 

[Here ](https://github.com/greiman/SdFs/tree/master/examples/ExFatLogger)is an example application using exFAT preallocation.

mjeanrichard, 2018-11-06:
Ok, thanks, I will keep an eye on that new lib.

closed 2018-11-06 by mjeanrichard