greiman/SdFat #118: Ejecting an SD card, more of a question really

GrahamJB, 2018-12-10:
My application allows the user to change SD cards, or even to run without one. Currently for eject I am closing all open files, saying it is safe to remove and using a local flag to determine if a card is present or not, After inserting a new card I call SdFat sdCard.begin again and reset the flag.

Secondly after an 'unexpected' error, not for example reading past the end of a file... I want to force an eject of the card and then ask for a new one (could be the same card, I have no control over that) and reinitialise it.

So is there a simple 'eject' method or a destructor for the sdCard which I am missing.

Regards
Graham Bland

greiman, 2018-12-10:
There is not a destructor for an SD card since I can't close all files open for write.

In any case try to close all open files.  If a file is open for write and close fails, there may be a corrupt file system.  If the file is open for read all should be OK even if close fails.

ask for the next SD and call begin()

GrahamJB, 2018-12-10:
Thank you.

One supplemental question, are there any memory issues with repeated calls to begin().

Regards

Graham Bland

greiman, 2018-12-10:
SdFat does not use dynamic memory so calling sd.begin() does not cause a change in memory use.

File classes also use no dynamic memory.

thijstriemstra, 2020-11-09:
Is there any way to detect when an SD-card is ejected @greiman?

greiman, 2020-11-10:
The only sure way is to use an SD module that allows a connection to the card detect switch.  [Here ](https://www.adafruit.com/product/254)is an example, see the CD pin.