greiman/SdFat #119: Endless loop in FatVolume::allocContiguous

mjeanrichard, 2018-12-16:
HI,

I (sometimes?) get an endless loop in when trying to ``createContiguous()``. 
The loop is in ``FatVolume::allocContiguous()``. The variable ``startCluster`` has a value of 1. Therefore the abort condition on line 157 (``if (startCluster == endCluster)``) is never true since ``endCluster`` is initialized to 2.

Am I missing something here? I assume that there is not enough space on the Card but this should not reult in an endless loop.

Thanks for oyur help!

greiman, 2018-12-17:
You are correct, there is a bug in how m_allocSearchStart  is used that can cause an infinite loop.

I am working on a fix.

mjeanrichard, 2018-12-17:
Cool, thanks!

greiman, 2018-12-20:
Try the latest version.  

I modified the allocation algorithms since I found additional problems.

mjeanrichard, 2018-12-21:
Thanks a lot fot the fast fix!

closed 2018-12-21 by mjeanrichard