greiman/SdFat #123: special character filename problem

mycroft2k, 2019-01-07:
with "file.printName(&Serial);"
give me '02 - H?hle.mp3'
"file.getSFN(shortname);"
Serial.println(shortname);
give me '02-H▒H~1.MP3'

all files with special charters are not working can you fix it? 

Filename on Windows PC/Unix
Long Name: '02 - Höhle.mp3'
Short 8.3 Name: '02-HÖH~1.MP3'

greiman, 2019-01-08:
SdFat only allows ASCII 7-bit characters and only the subset allowed by FAT16/FAT32.

If you create files on another computer with extended characters, they will not be usable with SdFat.

Here is the test for valid characters when a fie is created on SdFat.
```
inline bool lfnLegalChar(char c) {
  if (c == '/' || c == '\\' || c == '"' || c == '*' ||
      c == ':' || c == '<' || c == '>' || c == '?' || c == '|') {
    return false;
  }
  return 0X1F < c && c < 0X7F;
}
```
The current version SdFat will never support code pages or Unicode.  The mapping of code pages or Unicode to short filenames requires too much memory for small Arduinos.

I have implemented Unicode in exFat  for the [SdFs](https://github.com/greiman/SdFs) library but it seems few people use the library.

mycroft2k, 2019-01-11:
I have a SAMD21 microcontroller and have a lot of RAM ;-)
have now rewritten my lib for the vs1053 with openNext
and string.startsWith compare the first two signs this works for me

closed 2019-01-11 by mycroft2k