greiman/SdFat #124: STM32_Core_Arduino SDIO support

BennehBoy, 2019-01-15:
Would it be possible to add support for the SDIO HAL implementation within STM32_Core_Arduino  being developed by STM? -> https://github.com/stm32duino/Arduino_Core_STM32

There is a native SDIO implementation but it is very 'spartan' and relies upon FatFS which has very limited output capabilities -> https://github.com/stm32duino/STM32SD

This would make projects intended for multiple board/core support a lot simpler.

greiman, 2019-01-15:
I am working on a new version of SdFat that supports exFat and has a better structure for adding SDIO to more architectures.  I will look at adding STM32 SDIO.

I hope to release a first beta in a few weeks but STM32 SDIO would be later.

BennehBoy, 2019-01-15:
Great stuff, I'll keep an eye out for it.  

As a starter that may help, there is a HAL based patch up of SDFAT that was made for the STM32GENERIC core -> https://github.com/danieleff/STM32GENERIC

That core is now unsupported however,  and another person managed to get the sdfat patch working on Arduino_Core_STM32 by adding the appropriate pin initialisation, see here -> https://github.com/lucky01/SdFat