greiman/SdFat #129: Working around the delays while the SD buffer is being written to permanent memory

RipVW, 2019-02-02:
Greetings,

Am using an Adafruit Adalogger (ARM Cortex MO @ 48MHz.) to record 3-axis accelerometer data.  My script can record >2800 data sets per second right up to the point where the SD card's buffer is full and takes several ms to transfer the data. 

It wouldn't be so bad if the script could send a flush() and then continue to buffer data until the SD card was ready, but it appears that everything comes to a halt.  Obviously you've figured out how to deal with this in your LowLatencyLogger program.  Rather than just downloading and using your program I'd like to understand how it works.  Have read though your program a few times but can't seem to grasp the big picture.  Do you have a more prose like description of how your program keeps reading the accelerometer while the SD card is tranferring its buffer?

Many thanks for your time,

John

greiman, 2019-02-02:
SD cards can take up to 200 ms for a write.  Many are faster but there will always be long write times.

SD cards pause to do flash wear leveling.  This requires copying 128KB or more from an overused location to a new location.

Here are two ways to deal with this problem.  

For very fast logging, capture data in an interrupt routine and buffer it for the background.  The AnalogBinLogger example uses this method.

The second way is to check for SD busy and buffer data until the SD is not busy.  This method limits rates since SD writes are 512 byte sectors. LowLatencyLogger uses this method.

I can't help you more since I don't have time for all the people who want programming help.  You need to read a real-time programming book.

RipVW, 2019-02-03:
Thank you for the kindness of a prompt reply to a question you must have heard a jillion times.  I was trying to implement the strategy used in LowLatencyLogger but my attempts to check for SD busy all failed.  Will spend more time studying your program.

Have looked for real-time programming books and found this one:
https://smile.amazon.com/Embedded-Systems-Real-Time-Interfacing-Microcontrollers/dp/1463590156/ref=cm_cr_arp_d_product_top?ie=UTF8

Are there any books you'd recommend?

Thanks again for your time,
John

RipVW, 2019-02-28:
Have been pursuing your strategy for the AnalogBinLogger with a LIS3DH accelerometer that produces 8-bit x,y,z readings at 5 kHZ.  Using interrupts, structs, and a crude circular buffer my script can keep up with the readings and recover from being at least 1250 readings (250 mSec) behind.  My question is how often does a 200+ mSec delay due to wear leveling (or some other event) occur?  Am wondering if the script can recover from the first wear leveling event before the next one occurs.

Many thanks for your encouragement and time early in this adventure.  It's been a great learning experience.

John

greiman, 2019-02-28:
Modern high quality cards seldom have write latencies of 100 ms.

If you use a freshly formatted card you will seldom have long events close together.  Areas mapped for wear leveling are very large, many MB.

You should not have a problem.

RipVW, 2019-02-28:
Hi Bill,  Thank you for the prompt response.  Even though I've put in a lot of effort and improved my programming skills, I know the real reason for getting this far is the luxury of the fast processor with abundant memory.  My script would never keep up on an Arduino Mega.  Still lots to learn.