greiman/SdFat #130: call of overloaded 'print(uint64_t&)' is ambiguous

IeiuniumLux, 2019-02-02:
I'm trying to log an [uint64_t timestamp](https://github.com/mavlink/c_library_v2/blob/420c60ee5a927b905c3eb04f16e8d2dd7b57043c/common/mavlink_msg_camera_trigger.h#L7) value, but I' getting this error when I compile the code.

call of overloaded 'print(uint64_t&)' is ambiguous

Does the SDFat print() method support 64-bit values?

greiman, 2019-02-03:
The Arduino Print class does not support 64-bit integers and Print is the base class for files.

I wish Arduino would add member functions for int64_t and uint64_t to Print.  Print should be the base class for all calls to print and println.

greiman, 2019-02-03:
Here are some functions for printing 64 integers.
```
size_t print64(Print* pr, uint64_t n) {
  char buf[21];
  char *str = &buf[sizeof(buf) - 1];
  *str = '\0';
  do {
    uint64_t m = n;
    n /= 10;
    *--str = m - 10*n + '0';
  } while (n);
  pr->print(str);
}
size_t print64(Print* pr, int64_t n) {
  size_t s = 0;
  if (n < 0) {
    n = -n;
    s = pr->print('-');
  }  return s + print64(pr, (uint64_t)n);
}
size_t println64(Print* pr, int64_t n) {
  return print64(pr, n) + pr->println();
}
size_t println64(Print* pr, uint64_t n) {
  return print64(pr, n) + pr->println();
}

int64_t i64 = -9876543210123;
uint64_t u64 = 1234567890987;
void setup() {
  Serial.begin(9600);
  println64(&Serial, i64);
  println64(&Serial, u64);
}
void loop() {
}
```
Output:
```
-9876543210123
1234567890987
```

Do something like this.
```
File file;
// ....
uint64_t n;
// ...
print64(&file, n);
```

IeiuniumLux, 2019-02-03:
The size_t print64(Print* pr, uint64_t n) function did it for me. Thanks Bill.

closed 2019-02-03 by IeiuniumLux