greiman/SdFat #131: bin file to csv convertion on pc for LowLatencyLogger.ino on pc

jagdeepsingh02, 2019-02-05:
Hello sir, is it any way to convert the .bin files to .csv on pc beacuse it takes hours to convert on the arduino board

greiman, 2019-02-05:
Yes you can convert the file on a PC.  You will need to write a PC program.

[Here ](https://github.com/greiman/SdFat/tree/master/extras/AnalogBinLoggerExtras/bintocsv)is an example PC program for the AnalogBinLogger example.

You will need to modify the above example for the way you are using LowLatencyLogger.

You probably can use the free Microsoft Visual Studio to build the program.

jagdeepsingh02, 2019-02-05:
Hello sir I tried but i'm not a pro programmer could you please give some
advices ?
I successfully compiled and run the program from command prompt an I was
able to open bin file and create a CSV file but not able to read the data
and print on the CSV file

greiman, 2019-02-06:
I can't help you write programs.  You don't need to be a "pro programmer".

You can just modify the bin to csv Arduino code in LowLatencyLogger to run on the PC.

jagdeepsingh02, 2019-02-06:
Ok sir, i Will start again and ask you if I need some help.
Thanks again for your help

greiman, 2019-02-08:
You can use these three files to build a PC program that will convert bin files from the LowLatencyLogger example to csv.

You will need to edit these if you modified the example.

It is a console program with two arguments:

LowLatencyBin2Csv binFile csvFile

Here are thr three files.

UserTypes.h:
```
#ifndef UserTypes_h
#define UserTypes_h
#include <stdint.h>
#include <stdlib.h>
#include <stdio.h>
// User data types.  Modify for your data items.
#define FILE_BASE_NAME "adc4pin"
const uint8_t ADC_DIM = 4;
struct data_t {
	uint32_t time;
	uint16_t adc[ADC_DIM];
};
void printData(FILE* pr, data_t* data);
void printHeader(FILE* pr);
#endif  // UserTypes_h#pragma once
```
LowLatencyBin2Csv.cpp : 
```
// LowLatencyBin2Csv.cpp : This file contains the 'main' function. Program execution begins and ends there.
//
//  For Visual studio add pch.h or turn off pre-compiled headers in the project settings.

#include "UserTypes.h"
// Number of data records in a block.
const uint16_t DATA_DIM = (512 - 4) / sizeof(data_t);

//Compute fill so block size is 512 bytes.  FILL_DIM may be zero.
const uint16_t FILL_DIM = 512 - 4 - DATA_DIM * sizeof(data_t);

struct block_t {
	uint16_t count;
	uint16_t overrun;
	data_t data[DATA_DIM];
	uint8_t fill[FILL_DIM];
};

int main(int argc, char* argv[]) {
	FILE *binFile;
	FILE *csvFile;
	errno_t err;
	block_t block;
	if (argc != 3) {
		printf("missing arguments:\n");
		printf("%s binFile csvFile\n", argv[0]);
		return 0;
	}
	err = fopen_s(&binFile, argv[1], "rb");
	if (err) {
		printf("open failed for %s\n", argv[1]);
		return 0;
	}
	err = fopen_s(&csvFile, argv[2], "w");
	if (err) {
		printf("open failed for %s\n", argv[2]);
		return 0;
	}
	printHeader(csvFile);
	while (!feof(binFile)) {
		if (fread(&block, sizeof(block), 1, binFile) != 1) break;
		uint16_t i;
		if (block.count == 0 || block.count > DATA_DIM) {
			break;
		}
		if (block.overrun) {
			fprintf(csvFile, "OVERRUN,%d", block.overrun);
		}
		for (i = 0; i < block.count; i++) {
			printData(csvFile, &block.data[i]);
		}
	}
	fclose(binFile);
	fclose(csvFile);
	return 0;
}
```
UserFunctions.cpp:
```
#include "UserTypes.h"
// User data functions.  Modify these functions for your data items.

// Start time for data
static uint32_t startMicros;

// Print a data record.
void printData(FILE* pr, data_t* data) {
	if (startMicros == 0) {
		startMicros = data->time;
	}
	fprintf(pr, "%d", data->time - startMicros);
	for (int i = 0; i < ADC_DIM; i++) {
		unsigned adc = data->adc[i];
		fprintf(pr, ",%d", adc);
	}
	fprintf(pr, "\n");
}

// Print data header.
void printHeader(FILE* pr) {
	startMicros = 0;
	fprintf(pr, "micros");
	for (int i = 0; i < ADC_DIM; i++) {
		fprintf(pr, ",adc%d", i);
	}
	fprintf(pr, "\n");
}
```

jagdeepsingh02, 2019-02-08:
thank you very much sir, it's very helpful
Thanks again and have a good day sir

ccfol, 2019-04-29:
Hi @jagdeepsingh02 I have the same problem. 

Did you get to solve it ??

pauledd, 2019-07-28:
You could try this:
https://github.com/pauledd/bin2csv/tree/master
But it is just a noob beginner code without warranty to work flawlessly

ccfol, 2019-07-29:
Thanks a lot, I'm not really working anymore on this but thanks anyway.

Regards.

https://github.com/pauledd/bin2csv/tree/master