greiman/SdFat #134: Issue passing String to SD.exists()

linwp, 2019-03-14:
I'm trying to create daily files with a timestamp and want to test if the file already exists, so I can print a header if a new file is created.
For this purpose I've adapted the example from this issue https://github.com/greiman/SdFat/issues/17
to pass a previously built string to SD.open() and SD.exists.

This works perfectly for SD.open() but for SD.exists() I get following error during compilation.

---Error---
sketch_mar14b:16: error: no matching function for call to 'SdFat::exists(String&)'
   if (SD.exists(Filename)) {
                         ^
Arduino\Garbage\sketch_mar14b\sketch_mar14b.ino:16:25: note: candidate is:
In file included from Arduino\libraries\SdFat\src/FatLib/FatLib.h:28:0,
                 from Arduino\libraries\SdFat\src/SdFat.h:33,
                 from Arduino\Garbage\sketch_mar14b\sketch_mar14b.ino:2:
Arduino\libraries\SdFat\src/FatLib/FatFileSystem.h:186:8: note: bool FatFileSystem::exists(const char*)
   bool exists(const char* path) {
        ^
Arduino\libraries\SdFat\src/FatLib/FatFileSystem.h:186:8: note:   no known conversion for argument 1 from 'String' to 'const char*'

Using library SPI at version 1.0 in folder: AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI 
Using library SdFat at version 1.1.0 in folder: Arduino\libraries\SdFat 
exit status 1
no matching function for call to 'SdFat::exists(String&)'

---Code---
#include <SPI.h>
#include <SdFat.h>
#define SD_CS_PIN 10

SdFat SD;

File myFile;

void setup() {

  String Filename = "123456789012345678.bin";
  Serial.begin(9600);
  SD.begin(SD_CS_PIN);
  myFile = SD.open(Filename, O_RDWR | O_CREAT);
  myFile.close();
  if (SD.exists(Filename)) {
    Serial.println("yes");
  } else {
    Serial.println("no");
  }
}

void loop() {
}

greiman, 2019-03-14:
You can use the following workaround.

`if (SD.exists(Filename.c_str())) {`

I will add a new call for exists(String str).

linwp, 2019-03-18:
> 
> 
> You can use the following workaround.
> 
> `if (SD.exists(Filename.c_str())) {`
> 
> I will add a new call for exists(String str).

Thanks for the fast reply. Fix works like a charm! The whole library actually. Took me a while to figure out how to work with it (Didn't see the HTML doku for quite a while).
Thanks for all the work!

closed 2019-03-18 by linwp