greiman/SdFat #135: AnalogBinLogger and RTC (Wire) problem

hapecko, 2019-03-19:
Hi,
I’m building fast analog recorder, AnalogBinLogger is perfect for the job. I just need to add RTC stamp. I’m trying to do that by creating another log file after closing bin file with the most important parameters of the bin file and RTC stamp. I don’t need to get the time from RTC each time I’m closing bin file, the real time of arduino start is sufficient.
The problem is, each time I try to use I2C (Wire library), I have problem with setting ADC clock (with RTClib) or I get stuck with "createContiguous failed" (with Time library) when trying to record data.
As far as I know, it is connected with initializing I2C using Wire library. I’m using Arduino nano ATmega328p and DS1307.
I’d grateful or all ideas how to go round this problem.

greiman, 2019-03-19:
It may be that Wire plus the RTC library uses enough extra RAM to cause RTClib or SdFat to fail in createContiguous.

Are you using an Uno?

hapecko, 2019-03-19:
I have tried Uno and nano, both with ATmega328. The result is always the same.

greiman, 2019-03-19:
They all have very limited RAM.  How are you using RTClib ?

hapecko, 2019-03-19:
I use Wire.h and RTClib.h. Then I call

rtc.begin()

and read current date and time in setup by 

DateTime now = rtc.now();

greiman, 2019-03-19:
I tried it on my Uno and had no problem.

I added these things to actually time stamp the files.

Callback function.
```
RTC_DS1307 rtc;
//------------------------------------------------------------------------------
// call back for file timestamps
void dateTime(uint16_t* date, uint16_t* time) {
  DateTime now = rtc.now();

  // return date using FAT_DATE macro to format fields
  *date = FAT_DATE(now.year(), now.month(), now.day());

  // return time using FAT_TIME macro to format fields
  *time = FAT_TIME(now.hour(), now.minute(), now.second());
}
```
Added to setup
```
  Wire.begin();
  rtc.begin();
  SdFile::dateTimeCallback(dateTime);
```
The files get a timestamp and I can do several logs without failure.

hapecko, 2019-03-20:
Thank you for your help, it is working now. The problem was in additional log file which went over RAM limit.

closed 2019-03-20 by hapecko