greiman/SdFat #138: Compatible error with NeoGps after the new SdFat version

SrBrahma, 2019-04-14:
Just updated the SdFat, and a new error shows up on my PlatformIO.

[image]

Looks like the new "#include <fcntl.h>" conflits with the constants of the NeoGPS lib, as far I understood.

greiman, 2019-04-14:
fcntl.h is a OS system file.  I include it to avoid problems with other libraries/systems that use fcntl.h.

The problem should be fixed in NeoGPS.

You can try editing SdFatConfig.h to prevent use of fcntl.h .  Edit this section:
```
//------------------------------------------------------------------------------
/** If the symbol USE_FCNTL_H is nonzero, open flags for access modes O_RDONLY,
 * O_WRONLY, O_RDWR and the open modifiers O_APPEND, O_CREAT, O_EXCL, O_SYNC
 * will be defined by including the system file fcntl.h.
 */
#if defined(__AVR__)
// AVR fcntl.h does not define open flags.
#define USE_FCNTL_H 0
#elif defined(PLATFORM_ID)
// Particle boards - use fcntl.h.
#define USE_FCNTL_H 1
#elif defined(__arm__)
// ARM gcc defines open flags.
#define USE_FCNTL_H 1
#else  // defined(__AVR__)
#define USE_FCNTL_H 0
#endif  // defined(__AVR__)
```

SrBrahma, 2019-04-14:
Yeap.

#elif defined(__arm__)
// ARM gcc defines open flags.
#define USE_FCNTL_H 0

Making it 0 stopped the problem, as I also previously thought. I will open an issue on NeoGPS about this. However, I don't think they will update soon, as the last update was almost 1 year ago. Maybe you should warn the users about this conflict.

And, thank you very much for this lib! You did an awesome work!

greiman, 2019-04-15:
> Maybe you should warn the users about this conflict.

Unfortunately the number of conflicts between Arduino libraries is almost unlimited since so I have given up on warning many years ago.

Too bad the Arduino build process throws all libraries and code into one giant name space.