greiman/SdFat #139: problem working with 3 spi devices at same time with arduino due and sdfat library

MaxBoyman, 2019-04-14:
Hey,
Im working on a project with arduino Due, FT810 Display chip and two SD Cards one for FT810 image loading and another to save files as datalogger so there are 3 spi devices. 
1- FT810    cs pin=10  miso=12  mosi=11  sck=13
2- its SD Card    cs pin=5  miso=12  mosi=11  sck=13
3- second SD card    cs pin 53 miso  miso,mosi,sck=spi socket of due board
  
im using Sdfat library for accessing second sd card and for working with FT810 the library is Gameduino GD3. the problem is that when i try two use these two sd cards together with ft810 its not posibble.
the ft810 and its related sd card is working properly but i cannot use second sd card.
i tried to use sdfat library TwoCards.ino to use two sd cards but it shows errors:

when sd1 cs pin is 53
and sd2 cs pin is 5
errorCode: 0X30,0X1F

viseversa
when sd2 cs pin is 53
errorCode: 0X20,0X0

help me please its urgent...

greiman, 2019-04-14:
I have no problem with using the Due ICSP socket for SPI and pin 53 or 5 for CS.

I can't test with the FT810 display.

Here is output from the QuickStart example.

> SPI pins:
> MISO: 74
> MOSI: 75
> SCK:  76
> SS:   10
> 
> Be sure to edit DISABLE_CHIP_SELECT if you have
> a second SPI device.  For example, with the Ethernet
> shield, DISABLE_CHIP_SELECT should be set to 10
> to disable the Ethernet controller.
> 
> SD chip select is the key hardware option.
> Common values are:
> Arduino Ethernet shield, pin 4
> Sparkfun SD shield, pin 8
> Adafruit SD shields and modules, pin 10
> 
> Enter the chip select pin number: 53
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CHIP_SELECT to disable another device.
> 
> Card successfully initialized.
> 
> Card size: 31915 MB (MB = 1,000,000 bytes)
> 
> Volume is FAT32, Cluster size (bytes): 32768
> 
> Files found (date time size name):
> 1980-00-00 00:00:00    8388608 TeensyDemo.bin
> 
> Success!  Type any character to restart.
> 
> Restarting
> 
> Enter the chip select pin number: 5
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CHIP_SELECT to disable another device.
> 
> Card successfully initialized.
> 
> Card size: 31915 MB (MB = 1,000,000 bytes)
> 
> Volume is FAT32, Cluster size (bytes): 32768
> 
> Files found (date time size name):
> 1980-00-00 00:00:00    8388608 TeensyDemo.bin
> 
> Success!  Type any character to restart.

MaxBoyman, 2019-04-14:
hey, i tried till now and finally i solved the problem using software spi. now i can use my second sdcard in addition to ft810 and it's sdcard. i hope there will be no further problem with software spi. 
but about your answer, the sd card is working with each of those three cs pins when there is no other sd card and ft810 connected devices. but when i use spi pins for ft810 and sd card my second sd card could not be found by the microcontroller. i tried to use "two cards.ino" to found the solution if any, but there were no differences. 
what i guess is:
1- there may be a problem with sdfat library which can not disable other spi devices before trying to connect with them.
2- as i expected sdfat should define another spi speed, spi data order and spi data mode for new sd card is it possible that two libraries which is using spi pins interferring on defenition of these settings?

greiman, 2019-04-15:
> 1- there may be a problem with sdfat library which can not disable other spi devices before trying to connect with the sd card i need.
> 2- as i know sdfat should define another spi speed, spi data order and spi data mode for new sd card is it possible that two libraries which is using spi pins interferring on defenition of these settings?

1 When you use multiple SPI devices you should disable all by setting chip select high for all devices before initializing any device.  Libraries for SPI devices can not detect other devices that may interfere.

Add this code for each device with the possible exception of the first device to be initialized. 
```
  pinMode(CS_PIN_DEV_N, OUTPUT);
  digitalWrite(CS_PIN_DEV_N, HIGH);
```

2.  Every library using SPI must use SPI.beginTransaction(MY_SPI_SETTINGS) to select it's SPI options.

You can have any number of SPI devices as long as the there is not too much load on the SPI bus and all obey the SPI protocol using chip select and SPI.beginTransaction()/SPI.endTransaction().

It appears that the ft810 has a limit of 30 MHz SPI clock.  I use a higher rate for SdFat but that should not matter.  You could try a lower SPI rate.  This will also handle SPI bus loading problems due to poor wiring or too much capacitance.

```
#define SPI_SPEED SD_SCK_MHZ(30)

  if(!sd.begin(chipSelect, SPI_SPEED)) {
    // handle error
  }
```

PBudmark, 2019-07-21:
Late comment but still might be valuable

When initially driving ChipSelect-pins high, the following sequence is preferred (at least on Arduino)

`    digitalWrite(CS_PIN_DEV_N, HIGH); // Enables INPUT_PULLUP `
`    pinMode(CS_PIN_DEV_N, OUTPUT);    // Switches to Output, HIGH without intermediate LOW`

The reason is that the pin is INPUT after initialization (Hi-res), and switching to output directly sets the pin to output LOW and the chip will be selected and might connect to MISO and start looking for SCLK
The digitalWrite(..,HIGH) then unselects the device, but this takes up to 4uS on 16 MHz Arduino

To ensure ChipSelect is not floating while Arduino power up, a pullup to Vcc is useful, but the preferred sequence is still needed.

humza-13, 2020-07-06:
I am facing a smiler issue i am using 2 spi modules (6 pin oled and a heart rate sensor )  i am trying to connect them with arduino nano but some how only 1 of them work at a time can any one help me out ?