greiman/SdFat #140: NODEMCU CAN'T ACCESS SD CARD

kabilramar, 2019-05-02:
I connected SD card and nodemcu in the following manner

CLK   ->   D5
MISO ->  D6
MOSI ->  D7
CS      -> D6
GND   -> GND
VCC    -> Vin

I got the following error as Can't access SD card. Do not reformat.
SD errorCode: 0X50,0X0

What may be the reason to get this error (If I run any program in SDFAT library  except(Quickstart)I got this error )

kabilramar, 2019-05-02:
I am using ESP8266 core 2.5 and sdfat library version 1.0.7

greiman, 2019-05-02:
It's a read error at line 504 of SdSpiCard.cpp.

Quickstart uses a slow, 4 MHz, SPI rate so you may need to use a slower SPI rate.  

Here is the Quickstart code:

```
// Test with reduced SPI speed for breadboards.  SD_SCK_MHZ(4) will select
// the highest speed supported by the board that is not over 4 MHz.
// Change SPI_SPEED to SD_SCK_MHZ(50) for best performance.
#define SPI_SPEED SD_SCK_MHZ(4)
```

```
  if (!sd.begin(chipSelect, SPI_SPEED)) {
    if (sd.card()->errorCode()) {
    // ...
```

raghuram-bala, 2019-05-02:
Can I use the above code in other sdfat library examples. Will it work

greiman, 2019-05-02:
you can always select an SPI rate in sd.begin().

kabilramar, 2019-05-03:
How to add SPI rate in NODEMCU 

SPI rate in nodemcu is spi.clockFrequency(4000000) ????   Whether it is correcr.

kabilramar, 2019-05-03:
Got it Thank u sir

closed 2019-05-03 by kabilramar