greiman/SdFat #141: Problem with SD.begin() in SdFat while arduino SD working

Junikorn, 2019-05-09:
Working on SD Web server on ESP8266 (Wemos mini PRO) i encountered issue with card initialization. I use SS (15 in my chip) for chip select. SdFatInfo example from this library shows:
```
Card type: SDHC

Manufacturer ID: 0X0
OEM ID: 
Product: 
Version: 0.0
Serial number: 0XC8030000
Manufacturing date: 1/2004

cardSize: 0.00 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: false
OCR: 0XC0FF8000

SD Partition Table
part,boot,type,start,length
1,0X0,0X4,0,4982784
2,0X0,0X0,0,0
3,0X0,0X0,0,0
4,0X0,0X0,0,0
error: 
File System initialization failed.
``` 
while Arduino SD example shows
```
Card type:         SDHC
Clusters:          488416
Blocks x Cluster:  64
Total Blocks:      31258624

Volume type is:    FAT32
Volume size (Kb):  15629312
Volume size (Mb):  15263
Volume size (Gb):  14.91

Files found on the card (name, date and size in bytes): ... [Lists accurately whole card]
```
My usages of SD Arduino library work perfectly and I can't seem to initialize with this library.
I had similar issue with other sd cards. I have put 100kOhm pull up on MISO just to be sure but nothing changed. What other measures can I take to resolve this problem? Sorry for posting this as an issue (especially considering existence of #122) but I'am lost and desperately need long names feature that this library provides.
Kind reagrds and thanks for help

greiman, 2019-05-10:
I can't help much since I don't have your hardware.  the Arduino SD library uses a very old, more than 10 years, version of SdFat.  It uses a slow SPI clock rate.

Your card or wiring and SD socket/module may not support 50 MHz.

Try editing this line and changing the 50 MHz to a slower rate like the 4 MHz used in the SD.h example.

```
  // Initialize at the highest speed supported by the board that is
  // not over 50 MHz. Try a lower speed if SPI errors occur.
  if (!sd.cardBegin(SD_CHIP_SELECT, SD_SCK_MHZ(50))) {
    sdErrorMsg("cardBegin failed");
    return;
  }
```

Junikorn, 2019-05-12:
Lowering SPI speed to half helped. Should have dived into code more deeply instead of posting the issue. Thank you very much for assistance. Closing the ticket

closed 2019-05-12 by Junikorn