greiman/SdFat #143: equation for card capacity should be 512 not 1024 for version 2 cards

jontio, 2019-06-14:
Hi,

Looking at the `inline uint32_t sdCardCapacity(csd_t* csd) ` function in `src/SdCard/SdInfo.h`

We have...

```
inline uint32_t sdCardCapacity(csd_t* csd) {
  if (csd->v1.csd_ver == 0) {
    uint8_t read_bl_len = csd->v1.read_bl_len;
    uint16_t c_size = (csd->v1.c_size_high << 10)
                      | (csd->v1.c_size_mid << 2) | csd->v1.c_size_low;
    uint8_t c_size_mult = (csd->v1.c_size_mult_high << 1)
                          | csd->v1.c_size_mult_low;
    return (uint32_t)(c_size + 1) << (c_size_mult + read_bl_len - 7);
  } else if (csd->v2.csd_ver == 1) {
    uint32_t c_size = 0X10000L * csd->v2.c_size_high + 0X100L
                      * (uint32_t)csd->v2.c_size_mid + csd->v2.c_size_low;
    return (c_size + 1) << 10;
  } else {
    return 0;
  }
}
```
`return (c_size + 1) << 10;` should read `return (c_size + 1) << 9;` as the simplified specifications page 172 say `memory capacity = (C_SIZE+1)*512KBytes`

Cheers,
Jonti

greiman, 2019-06-14:
C_SIZE is a 22-bit field that specifies a a 32-bit block count.  The low 10-bits are zero so you shift by 10 to get block count then multiply by 512 to get block count.

The simplified spec is not very clear.  512Kbytes is 2 << 19 or 524288.

jontio, 2019-06-14:
Ops. I see. Thanks for the info. I should have read the declaration first. The function returns the number of 512byte blocks and not KBytes. That makes sense as I see for reading data (say CMD18) you specify the block number as the argument.

I'd been writing some low level stuff to learn about the how SD spi works and I had just managed to read the CSD register and got back C_SIZE=30386. I wanted to know what size the card was; and (30386+1)*512 on the calculator looked more familiar to me (about 16 with lots of zeros).

So by "KBytes" they mean 1024 bytes. When I see KB/kB I am always wondering if it's 1000 or 1024; they say usually KB=1024B and kB=1000B but I know it catches a lot of people out as 1000 and 1024 aren't to dissimilar.

So...

(C_SIZE+1)*512*1024 == number of bytes

512 bytes == 1 block

(C_SIZE+1)*1024 == number of blocks

Sorry for my mistake.

Cheers,
Jonti

closed 2019-06-14 by jontio

greiman, 2019-06-15:
You may want to look at [SdFat-beta](https://github.com/greiman/SdFat-beta).  It is a major new version that supports exFAT.