greiman/SdFat #144: Read list of file in directory

emid-alipour, 2019-07-01:
Hi

how can i to getting list of dierctory and read all file in directory?

Thanks Lot.

greiman, 2019-07-01:
You can use openNext() to open and read every file in a directory.

The [openNext](https://github.com/greiman/SdFat/blob/master/examples/OpenNext/OpenNext.ino) example opens all files in a directory.  You need to add code to read each file.

emid-alipour, 2019-07-08:
im using your code in loop ever 5 Secound,
but its not working so
my code:

`/*
 * Print size, modify date/time, and name for all files in root.
 */
#include <SPI.h>
#include "SdFat.h"

// SD default chip select pin.
const uint8_t chipSelect = SS;

// file system object
SdFat sd;

SdFile root;
SdFile file;
//------------------------------------------------------------------------------
void setup() {
  Serial1.begin(115200);
  
  // Wait for USB Serial 
  while (!Serial1) {
    SysCall::yield();
  }
  
  Serial1.println("Type any character to start");
  while (!Serial1.available()) {
    SysCall::yield();
  }

  // Initialize at the highest speed supported by the board that is
  // not over 50 MHz. Try a lower speed if SPI errors occur.
  if (!sd.begin(chipSelect, SD_SCK_MHZ(50))) {
    sd.initErrorHalt();
  }
  if (!root.open("/")) {
    sd.errorHalt("open root failed");
  }
  // Open next file in root.
  // Warning, openNext starts at the current directory position
  // so a rewind of the directory may be required.
  while (file.openNext(&root, O_RDONLY)) {
    file.printFileSize(&Serial1);
    Serial1.write(' ');
    file.printModifyDateTime(&Serial1);
    Serial1.write(' ');
    file.printName(&Serial1);
    if (file.isDir()) {
      // Indicate a directory.
      Serial1.write('/');
    }
    Serial1.println();
    file.close();
  }
  if (root.getError()) {
    Serial1.println("openNext failed");
  } else {
    Serial1.println("Done!");
  }

  
  // Initialize at the highest speed supported by the board that is
  // not over 50 MHz. Try a lower speed if SPI errors occur.
  if (!sd.begin(chipSelect, SD_SCK_MHZ(50))) {
    Serial1.println("error");
    sd.initErrorHalt();
  }

  
}
//------------------------------------------------------------------------------
void loop() {
  
  Serial1.println("check");

  if (!root.open("/")) {
    sd.errorHalt("open root failed");
    Serial1.println("open root failed");
  }
  else
  {
      Serial1.println("okkk");
    // Open next file in root.
    // Warning, openNext starts at the current directory position
    // so a rewind of the directory may be required.
    while (file.openNext(&root, O_RDONLY)) {
      file.printFileSize(&Serial1);
      Serial1.write(' ');
      file.printModifyDateTime(&Serial1);
      Serial1.write(' ');
      file.printName(&Serial1);
      if (file.isDir()) {
        // Indicate a directory.
        Serial1.write('/');
      }
      Serial1.println();
      file.close();
    }
    if (root.getError()) {
      Serial1.println("openNext failed");
    } else {
      Serial1.println("Done!");
    }
  }
  
  delay(5000);
  
  
  }`

and result in terminal it show only running for one time and it going to hang:

[image]

emid-alipour, 2019-07-10:
i get list file with file.printName(&Serial1); command, but this command only used for serial port, how can i to save filename in a char variable?

greiman, 2019-07-10:
use 
`bool FatFile::getName (char* name, size_t  size);`

You should read the html documentation located in Arduino/libraries/SdFat/extras/SdFat.html.

Emid-norton, 2019-07-10:
Thanks lot for your help :)