greiman/SdFat #146: Low capacity SD cards ?

yet-another-average-joe, 2019-07-21:
Hello !

I am discovering your library for a DIY STM32 project that doesn't require a huge capacity : a few KB at most...

Therefore, I dug into the crap, and found a 16MB Panasonic SD card (sixteen megabytes). You know, this kind of memory card that came with cameras back in the days... (I also attempted to insert an 8" floppy disk, but unfortunately, it didn't fit...)

I spent about 4 hours trying to figure out what I did wrong : "Initializing SD card...initialization failed!". I tried out SPI1 and SPI2, verified and re-verified everything, with FAT16 and FAT32 format. No success...

And then, I had an idea ! I inserted a 1GB SD card (Yes ! One GigaByte !). And it worked ! (I'm cheap)

Is there a lower capacity limit or whatever ? Could you add a warning somewhere ?

(just kidding)

Why this limit, and what is it ?

greiman, 2019-07-22:
The standard filesystem for small SD cards is FAT12.  By default SdFat supports only FAT16/FAT32 since FAT12 cards are no longer manufactured.  FAT12 requires over 500 bytes of flash on an Uno.

The SD standard specifies cards smaller than 128 MB are formatted FAT12.  Your 128 MB card is close to the FAT12/FAT16 boundary and must have been FAT12.

There are two ways to use small cards.

You can edit SdFatConfig.h at about line 147 and enable FAT12.
```
/**
 * Set FAT12_SUPPORT nonzero to enable use of FAT12 volumes.
 * FAT12 has not been well tested and requires additional flash.
 */
#define FAT12_SUPPORT 0  // <<<<< Set this to one.
```
You can use the the SdFormatter example which will format small cards FAT16 if they are 16 MB or larger.  This is nonstandard for small SD cards but is useful if you want a smaller cluster size.

yet-another-average-joe, 2019-07-22:
Thanks for the quick answer !

I completely forgot about FAT12/16 ! (30 years ago or so...)

Now the 16MB SD card works.

I tested the 128MB card formatted with gparted instead of Windows. Initialization fails with errorCode 0x43, errorData 0x05 for both FAT16 and FAT32, but it is readable/writable under Windows and Raspbian. No option for FAT12. 

And writing this, I just realized that the 128MB is NOT a SD, but a MMC.

greiman, 2019-07-22:
The error code is because the MMC card does not support a command I use to init an SD card.

I decided not to support MMC cards.  The MMC standard now is mostly used for high performance  soldered eMMC memory modules inside tablets, phones, not as removable media.

yet-another-average-joe, 2019-07-22:
Problem solved, questions answered, thank you again !

closed 2019-07-22 by yet-another-average-joe