greiman/SdFat #149: AnalogBinLogger record Button

pauledd, 2019-08-02:
Hi

I try to modify your AnalogBinLogger example so that I can push a button to start
recording the adc data and push it again to stop but I failed early in the overwhelming amount of code in your example.

I simply dont get it where to place a checkpoint for a digital Pin assigned to the button.
If I put it right after the beginning of the loop() function it runs only once. Is there a place where I can put it and gets executed continuously while serial output is not active? 
Any ideas?

greiman, 2019-08-04:
I don't customize examples.  Maybe someone else can help you.

mantoles, 2020-09-24:
@pauledd I was wondering if you ever got this working.  I know this was from last year but I am in the same boat using the code.  In fact I was curious to find out if you had to modify the code to read the analog pins correctly.  It is not very clear to me how we are making analog reads real time with the code since we only do an actual analog read once.  

// Read the first sample pin to init the ADC.
  analogRead(PIN_LIST[0]);

Very elegant code and appreciative of the effort but my runs only return 1023 for all analog values as if they are all stuck high even though my other program returns proper values.

greiman, 2020-09-24:
I checked the example.  I put a 1,000 Hz sine signal on pin 0 and the Arduino 3.3V power on pin1.  Here is a chart at 15,000 samples per second using an Uno. The time scale is in milliseconds.  It is a smooth excel plot so the sine does not show individual points. 
[image]

mantoles, 2020-09-24:
Much appreciate you following up on it.  I will try your example first and then use it to capture the event I am looking for which is a 4ms pulse at 5 volts amplitude .  Your program is more than capable of capturing this

mantoles, 2020-09-24:
Switched from a leonardo to a mega and getting excellent performance with this logger.  Thank you !!