greiman/SdFat #151: teensy 4 support

mr-stivo, 2019-09-04:
Any chance of teensy 4 support?

Trying to compile SdInfo:
/Users/stivo/Documents/Arduino/libraries/SdFat/src/SpiDriver/SdSpiTeensy3.cpp:28:21: fatal error: kinetis.h: No such file or directory

greiman, 2019-09-04:
I will add Teensy 4 SPI support to SdFat-beta.  I am very busy with other things now so it may be a while.  

SPI support will be first since it will be easy.  SDIO will be a bit later.

greiman, 2019-09-04:
I just tried deleting the include for kinetis.h.  It appears that SPI support works for Teensy 4.

It's at about line 28 of SdSpiTeensy3.cpp

```
#include "SdSpiDriver.h"
#if defined(__arm__) && defined(CORE_TEENSY)
// SPI definitions
//#include "kinetis.h"  <<----- Remove this line
```

I ran the bench example on a Teensy 4 and got these results.

```
Type any character to start
chipSelect: 10
FreeStack: -1605688
Type is FAT32
Card size: 32.01 GB (GB = 1E9 bytes)

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Version: 1.0
Serial number: 0X1550FE3
Manufacturing date: 10/2015

File size 5 MB
Buffer size 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
411.12,24090,1107,1244
413.68,21659,1106,1237

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1031.50,1409,482,496
1031.50,1411,487,496

Done
```

mr-stivo, 2019-09-04:
Thanks for the quick reply. I will give it a try today.

I saw in the PJRC forums that Teensyduino 1.48 may address the SdFat incompatibility ( along with some others). Hopefully we will see kinetis.h for the teensy4 and SDIO support will work.
https://forum.pjrc.com/threads/57216-Teensyduino-1-47-Released

Thanks again.

mr-stivo, 2019-09-05:
I made the changes but was not able to detect the sd card. I am using the pads on the underside of the teensy4 (SDIO/SPI2 - pins 34~39). May I ask how/where your SD card is connected?

Thanks again for the help.

greiman, 2019-09-05:
The pins on the bottom will not work for SPI.  Unlike the Teensy 3, a socket connected for SDIO will not be connected correctly for SPI.   I contacted Paul and he said there is no way to map the pins correctly.

I connected to the standard  SPI port.

With SdFat-beta, you can specify the port.  You should be able to use SPI1 but I don't see the pin for MISO1 on the bottom pinout sheet.

greiman, 2019-09-05:
looks like pin1 is SPI1 MISO1, pin 26 MOSI1, pin 27 SCK1.

https://forum.pjrc.com/threads/54711-Teensy-4-0-First-Beta-Test?p=193716&viewfull=1#post193716

mr-stivo, 2019-09-05:
Ohhh I see. I just assumed the SPI and SDIO pins were placed on that pad so I could use either. I didn't notice the SPI pins were wrong for an SD card.

I'll keep my eye on SdFat-beta for SDIO support.

Thank you so much for checking this out so quickly.

closed 2019-09-05 by mr-stivo