greiman/SdFat #153: Incorrect SD cards capacity

MyrmidonGitHub, 2019-09-20:
Hi!

It is my first time with SDFat, and I found something strange when I checked the capacity of two microSD cards with SDFat 1.1.0 and Arduino IDE 1.8.9.

I include a brief report of the data I got with two different cards, previously formatted with SD Card Formatter 5.0.1 of the SD Association:

I use Arduino Mega 

**********************************************************
Transcend 32GB card:
**********************************************************
arduino SDFat:    33555.48 * 1000000 =  33555480576 Bytes
arduino SD:        32756736 *      1024 =  33542897664 Bytes
Windows:                                                 33542897664 Bytes

arduino SDFat - arduino SD = 12582912 Bytes
arduino SDFat - Windows    = 12582912 Bytes

SDFat 1.1.0 SDFormatter cannot format this card:
16:17:54.789 -> Erasing
16:18:04.869 -> error: erase failed
16:18:04.869 -> SD error: 63,0

**********************************************************

**********************************************************
TDK        2GB card:
**********************************************************
arduino SDFat:      1973.42 * 1000000 =	 1973420032 Bytes
arduino SD:     1924960     *        1024 =  1971159040 Bytes
Windows:                                                  1971159040 Bytes

arduino SDFat - arduino SD =  2260992 Bytes
arduino SDFat - Windows    =  2260992 Bytes

SDFormatter can format this card, and after formatting, 
Windows says 1973092352 Bytes.

After re-formatting with SD Card Formatter, Windows says 
again 1971159040 Bytes.
**********************************************************

Is there any known issue about this ?

Thank you in advance

greiman, 2019-09-20:
SD.h is based on a version of SdFat that is over ten years old. The CardInfo example has been changed a number of times by Arduino over the years.  I can't say what it does now.

Window Capacity is not the card size it's the total clusterCount * clusterSize.

The only thing that will be consistent is total cluster count.  Free cluster count will vary if you mount a card on windows.  Windows may add or delete info from the 'System Volume Information' folder.

Here are results for a 32 GB Samsung card.

Windows reports:
```
Used space: 131,072 bytes
Free space: 31,998,214,144 bytes
Capacity: 31,998,345,216 bytes
```

I added this line to the [SdFat-beta](https://github.com/greiman/SdFat-beta) SdInfo example. 
`cout << F("totalSectors: ") << sd.card()->sectorCount() << endl;`

The SdFat SdInfo example reports:
```
cardSize: 32010.93 MB (MB = 1,000,000 bytes)
totalSectors: 62521344
sectorsPerCluster: 64
clusterCount:      976512
freeClusterCount:  976508
```
`CardSize = 512*totalSectors = 512*62521344 = 32,010,928,128 bytes`

Each cluster has 64*512 bytes.

To get capacity multiply clusterCount by cluster size.
`976512*64*522 = 31,998,345,216‬`

To get free space multiply freeClusterCount by cluster size.
`976508*64*522 = 31,998,214,144`

Free space = capacity - used space = 131,072.

In this case SdFat-beta and Windows agree.  SdFat 1.1.0 gets the same result.

MyrmidonGitHub, 2019-09-21:
Thank you very much for your fast response.

I understand all your calculations, but for my 2G card, SdInfo v1.0.1 shows:

02:29:21.483 -> 
02:29:21.483 -> SdFat version: 10100
02:29:21.483 -> 
02:29:21.483 -> Disabling SPI device on pin 10
02:29:21.483 -> 
02:29:21.483 -> Assuming the SD chip select pin is: 4
02:29:21.483 -> Edit SD_CHIP_SELECT to change the SD chip select pin.
02:29:21.483 -> 
02:29:21.483 -> type any character to start
02:29:22.323 -> 
02:29:22.323 -> init time: 2 ms
02:29:22.323 -> 
02:29:22.323 -> Card type: SD2
02:29:22.323 -> 
02:29:22.323 -> Manufacturer ID: 0X2
02:29:22.323 -> OEM ID: TM
02:29:22.323 -> Product: SA02G
02:29:22.323 -> Version: 0.3
02:29:22.323 -> Serial number: 0XBFAB7F15
02:29:22.323 -> Manufacturing date: 7/2011
02:29:22.323 -> 
02:29:22.323 -> cardSize: 1973.42 MB (MB = 1,000,000 bytes)
02:29:22.323 -> flashEraseSize: 128 blocks
02:29:22.323 -> eraseSingleBlock: true
02:29:22.323 -> OCR: 0X80FF8000
02:29:22.323 -> 
02:29:22.323 -> SD Partition Table
02:29:22.323 -> part,boot,type,start,length
02:29:22.323 -> 1,0X0,0X6,137,3850423
02:29:22.323 -> 2,0X0,0X0,0,0
02:29:22.363 -> 3,0X0,0X0,0,0
02:29:22.363 -> 4,0X0,0X0,0,0
02:29:22.363 -> 
02:29:22.363 -> Volume is FAT16
02:29:22.363 -> blocksPerCluster: 64
02:29:22.363 -> clusterCount: 60155
02:29:22.363 -> freeClusters: 60145
02:29:22.763 -> freeSpace: 1970.83 MB (MB = 1,000,000 bytes)
02:29:22.763 -> fatStartBlock: 138
02:29:22.763 -> fatCount: 2
02:29:22.763 -> blocksPerFat: 235
02:29:22.763 -> rootDirStart: 608
02:29:22.763 -> dataStartBlock: 640
02:29:22.763 -> 
02:29:22.763 -> type any character to start

I see:
 blocksPerCluster: 64
 clusterCount: 60155

So if I'm not wrong, the card size in bytes should be 60155 * 64 * 512 = 1971159040
but it says cardSize: 1973.42 MB (MB = 1,000,000 bytes) 

Sorry, I don't understand it, maybe cause it's 2:38 a.m.
Am I making any mistake?

Thank you again, and please forgive me if my questions seem too silly! :)

greiman, 2019-09-21:
> So if I'm not wrong, the card size in bytes should be 60155 * 64 * 512 = 1971159040

This is what Windows calls Capacity.  It is cluster count times cluster size.  It does not include FAT tables and other file structures.

Card Size includes all sectors so is larger than Capacity.  That's why card size is 1973.42 MB.

MyrmidonGitHub, 2019-09-21:
Oh, finally I understood!
and I forgot to congratulate you for your library!

The 32GB Kingston microSD (which in my first message I incorrectly identified as Transcend) can be formatted with the utility of the SD Association, but not with your SdFormatter example, in which it returns:

Erasing
error: erase failed
SD error: 63.0

Is it a known issue?

Thank you very much again.

greiman, 2019-09-21:
Try a quick format. with the SdFat SdFormatter.  By default, the SD Association formatter does a quick format.

It's rare for SdFormatter to fail with erase.  Mostly fake cards, which are very common, fail.  

Try the SD Association formatter with the erase option.  It may also say flash erase fails.

MyrmidonGitHub, 2019-09-21:
The SD Association formatter can format it with both the quick and the overwrite (full) options, but with your SdFormatter I get these results:

E - erase the card and skip formatting.:
16:57:44.862 -> Erasing
16:57:54.942 -> error: erase failed
16:57:54.942 -> SD error: 63,0

F - erase and then format the card. (recommended):
16:59:52.293 -> Card Size: 33555 MB, (MB = 1,000,000 bytes)
16:59:52.293 -> 
16:59:52.293 -> Erasing
17:00:02.375 -> error: erase failed
17:00:02.375 -> SD error: 63,0

Q - quick format the card without erase.:
17:01:14.516 -> Formatting
17:01:14.516 -> Blocks/Cluster: 64
17:01:14.516 -> FAT32
17:01:14.556 -> ...............................................................
17:01:27.717 -> Format done

As you pointed, I'm not sure if my card is fake. Its manufacturer id and oem id are from "AData" instead of "Kingston".
AData	0x00001d	AD	AData
Kingston	0x000041	42	Kingston

Maybe next time I buy an SD card I should carry an Arduino with LCD to check it..., but the declared capacity is correct.

Is the format algorithm manufacturer dependant?

What does you 63 error mean?

Thank you again.

greiman, 2019-09-21:
The format is not manufacturer dependent.  It is specified by card size.  Cards are built to optimize the standard format.  For example, file system structures are aligned on flash allocation boundaries.  Wear leveling is done by flash allocation unit so placing thing that don't change in their own flash allocation unit means they won't be copied due to wear leveling.

error 63 is SD_CARD_ERROR_ERASE_TIMEOUT. 

I don't understand since the timeout is 10 seconds or 10000 ms. That is for 262,144 sectors.
`const uint16_t SD_ERASE_TIMEOUT = 10000;`

Looks like it fails at once.

I timed the erase of a 32GB Samsung card and the entire card erased in under eight seconds. A 32GB SanDisk card erased in about four seconds. So each 262,144 sector section is erasing in about 2-4 ms.

MyrmidonGitHub, 2019-09-21:
As you can see in my previous text, the time marks show the 10s timeout in both the
E - erase the card and skip formatting.:
and
F - erase and then format the card. (recommended):

I must add this SD was empty (no file).

May I help you (and myself) to adjust the timeout or something else ?

Thank you again.

greiman, 2019-09-21:
It won't help to adjust the timeout.  The timeout is 10 seconds and there are about 240 erase calls for a 32GB card.  It would take more than 2,400 seconds.

Your card must hang.  That's possible and still have the SD formatter work.  Arduino use a different interface, SPI vs SDIO.

I have used more than 50 SD cards in tests and none have failed.  I only have one or two other reports of failures from users so I can't pursue the issue.

MyrmidonGitHub, 2019-09-21:
Well, don't worry. I assume my card is fake.

Thank you very much for your assistance and congratulations again for your magnificent work.

If you don't need anything else, I will close the thread.

closed 2019-09-21 by MyrmidonGitHub