greiman/SdFat #154: CS as a function?

kostuch, 2019-10-11:
Hi,
Not an issue, but rather question.
I have a device based on ESP8266 where I use I2C expander PCF8575 to have more I/O.
CS_PIN input of SDCard reader is connected to one of expander output. I can set/reset this pin calling functions in PCF library. But I can not use direct digitalWrite(CS_PIN), as this is pad on the external chip.
I wonder if it possible to hook an external function (via callback maybe?) to allow different initialization and CS manipulation.
I did it some time ago, but it was very dirty hack and I lost my sources during upgrade.

Regards

greiman, 2019-10-11:
I suspect you could make it work.

Be aware that I will soon release SdFat-beta as SdFat V 2.0.  It has different internals.

kostuch, 2019-10-12:
OK, so I'll wait for a new release, then I'll try to implement it in more elegant way than before.