greiman/SdFat #155: FatFileLFN with 8-bit ASCII support

madias123, 2019-10-12:
More a feature request than a bug...
Support for long filennames is only 7-bit ASCII. Most of the time there is no problem about that, unless you want to build an Audio/Video player with a display and you need to show the filenames written in most other languages than english...
At this time there is no chance to get  such special characters, neither the ASCII code, because they were converted to "?".
Maybe there is a chance to fix that?

Thank you
Matthias

madias123, 2019-10-12:
I've made some progress:
FatFileLFN.cpp:
Changes:
Line 78 to: name[k++] = c >= 0XFF ? '?' : c;
Line 537 to:       if (u > 0XFE) {
        u = '?';
With this change, the special characters are read. However, a file / directory can not be opened with this char array. (Is not necessary.)

greiman, 2019-10-12:
This version of SdFat of SdFat will soon be replaced by SdFat V 2.0 so only serious bugs will be fixed.

V 2.0 is now in beta.  I will look at extended support as an option.  The problem is mapping eight bit characters to the required short file names.  This is usually done with "code pages".  This would be an option that would increase the size too much for many Uno apps.

https://github.com/greiman/SdFat-beta

madias123, 2019-10-12:
Ok, I understand as more I grow into "Unicode FAT32". 
I'm using STM32(duino) (Roger Clark Core) so I know about the restrictions (and spit up)  of UNO (there is really no fun about this). 
I  solved my "problems" with my hack above, but going to use v2 in the future.

BTW: Really, just "thanks" for your work!