greiman/SdFat #163: Piping a String with ArduinoOutStream

kamocat, 2019-11-13:
Hi there,

I found a strange issue: When I try to use a String variable in a pipe, it prints a "1".
Here's a minimal example:
[https://gist.github.com/kamocat/f3caa98a0e6592877591e176217f1c5b]

I tried this on an Adafruit Feather m0. I imagine it's platform-independent.

greiman, 2019-11-13:
There is no support for String in sdios.h.  I am not sure why this compiles.

I tried to add a << operator for type String but there is no change.

Something in the String class is preventing my opetator<< function from working.

I made a note to try to find a solution in SdFat V2.

kamocat, 2019-11-13:
Ok. As a workaround, I can use the String::c_str() method, or store my object as a C string natively.

greiman, 2019-11-14:
Add this function to your sketch.

```
#include <sdios.h> // for ArduinoOutStream

// Add here until I have a better solution.
inline ostream &operator<< (ostream &os, const String &arg) {
  os << arg.c_str();
  return os;
}

ArduinoOutStream cout(Serial);

void setup() {
  Serial.begin(115200);
}

void loop() {
  String str1 = "Print OK, ";
  String str2 = "Pipe FAIL";
  Serial.print(str1);
  cout<<str2<<" (Wrong value piped)"<<endl;
  delay(1000);
}
```

I will put the fix in [SdFat V2](https://github.com/greiman/SdFat-beta).  I have frozen SdFat V1.  I will soon release the beta as V2.

greiman, 2019-11-18:
I added support to [SdFat V2 beta](https://github.com/greiman/SdFat-beta).

kamocat, 2019-11-23:
Wow, that was fast.
Thanks for your hard work!

closed 2019-11-23 by kamocat