greiman/SdFat #165: SdFat does not compile if variant does not SPI defined

sslupsky, 2019-11-15:
I have found an issue where if the board variant does not define SPI, then the compilation fails here: 

https://github.com/greiman/SdFat/blob/3b79f38cb554809a45a0ccd6d9d6752b6ad948c2/src/SpiDriver/SdSpiDriver.h#L38

Later on, there are other issues such as when SDCARD_SPI
In my situation, I have a couple board variants.  One with a hardware SPI and another without.  In the case I do not have a hardware SPI port, I would like to use the SoftSPI driver.  However, since SPI is not defined, it will not compile.

greiman, 2019-11-16:
SdFat V1 is frozen.  I will soon release [SdFat V2](https://github.com/greiman/SdFat-beta).

SdFat V2 has very limited support for Software SPI.  I don't plan to extend Software SPI support anytime soon.

sslupsky, 2019-11-17:
Nevertheless, if the variant does not specifically define "SPI" (uses another variable name for instance), the SdFat library will not compile.

greiman, 2019-11-17:
> SdFat V1 is frozen. I will soon release SdFat V2.
> 
> SdFat V2 has very limited support for Software SPI. I don't plan to extend Software SPI support anytime soon.

Sorry I was not clear.  This means SdFat V1 will not be modified and SofrwareSPI is a low priority for SdFat V2.

SoftwareSPI was added so SdFat could be used when there was a problem with interference between SdFat and another SPI device.  The assumption was that hardware SPI was available and SPI.h could be included.  

It is not simple to fix this problem since SdFat makes heavy use of the classes SPIClass and SPISettings which are defined and implemented in the Arduino SPI library.

I am working on solving a more general problem that will fix SPI not defined for SoftwareSPI.  I need to generalize SPI support so systems other than Arduino can use SdFat.  In fact I want to isolate all Arduino dependency so I can use SdFat with native ChibiOS and other RTOS systems. 

I will  release a first step in the future but I need to do a lot of testing since the changes impact lots of files.  Drivers for Due, SAMD, Teensy 3.x and 4.0, STM32, and ESP8266.

greiman, 2019-11-18:
I posted the latest [SdFat V2](https://github.com/greiman/SdFat-beta).

Software SPI may work for systems that do not have SPI.h.

I can't do a good test since I don't have hardware.

Try the [SoftwareSpi ](https://github.com/greiman/SdFat-beta/tree/master/examples/SoftwareSpi)example.

Edit SdFatConfig.h at about line 106 like this:
`
#define SPI_DRIVER_SELECT 2`