greiman/SdFat #167: problem with deleting a short lower-case file name

nealcrook, 2019-11-19:
First, thanks for creating and sharing and supporting this excellent library.

Now, my problem.. on Arduino (ATMEGA) configured for short file names:

#define USE_LONG_FILE_NAMES 0

I have a file "UPPER.TXT" and a file "lower.txt"

When I open the first file I can refer to it as "upper.txt" or as "UPPER.TXT"
When I list the directory, they appear as UPPER.TXT and LOWER.TXT

I think this is all correct, part of the madness of FAT. However, when I try to remove the second file I cannot remove it,

FatFile *working_dir;
FatFile handle;

SD.chdir("MYDIR", 1);
working_dir = SD.vwd();
if (!handle.remove(working_dir, "lower.txt")) {
  // fails
}
if (!handle.remove(working_dir, "LOWER.TXT")) {
  // fails
}

is this a bug or am I misusing the library

thanks,

Neal.

greiman, 2019-11-19:
I don't recall why the static function  `static bool FatFile::remove(FatFile* dirFile, const char* path);` exists.

To remove your file I would expect the following to work.

```
  SD.chdir("MYDIR", 1);
  if (!SD.remove("LOWER.TXT")) {
    // fails
  }
```

If the file ""lower.txt" was created as LFN, it won't be possible to remove it in SFN mode.

nealcrook, 2019-11-21:
Thanks for your comments. I tried SD.remove("LOWER.TXT") but it failed in the same way as handle.remove(working_dir, "LOWER.TXT")) -- but is a few bytes smaller. I assume that FatFile::remove() exists for consistency.

I guess the fundamental problem is that "lower.txt" *is* a "long file name" -- it wasn't obvious to me before that lower-case files fell into that category (and https://en.wikipedia.org/wiki/8.3_filename confirmed it).

regards,
Neal.

closed 2019-11-21 by nealcrook