greiman/SdFat #168: Writes to SD card causes another device on the SPI bus to read incorrectly.

JonRobert, 2019-11-23:
I have two devices on the SPI bus of an Arduino M0
  -  SD Card Breakout
  -  MAX31855k breakout,  Simple read only board.

My loop consists of:
 -  reading the MAX board
 - writing to the SD card
 - delay(2000)

The issue I've found is on the clock of the MAX board read, an extra leading pulse on the clock line occurs if I write to the SD card before reading the MAX. 

I also have seen this additional pulse on startup before the SD card has been written to.  However it does not appear on any read after that (if no SD card is read)

I suspect the read to the SD card changes an SPI setting.   If I use my imagination, I would guess the first read on the SPI has a clock setup issue which clears after the 1st read.
Adding in the SD card, my guess is, the SD card read changes the SPI settings causing the next MAX read to be like the first.

Any thoughts on how to resolve this issue?   Note, I tried the code without the MAX and the long pulse was not there???  but the delay from the MAX CS dropping to the first real clock pulse is the same a it was even with the added long clock pulse.

[image]

`/*
  SD card read/write + TypeK read

  Rev 03 tested successfully on Arduino M0
  Rev 04 Combined Basic_SD and Type K, Sharing SPI  <<<<<<===========================!!!!
         The goal of this program is to determine how to share the SPI between the SD Card
         which opens the SPI file from the SdFat library.
         And the M31855k which uses the "normal" SPI creation.
         Modified while to do / while.
*/

#include <SPI.h>
//#include <SD.h>
#include "SdFat.h"
SdFat SD;

// *** SD Card declarations **************************************************
// ***************************************************************************

const uint8_t SDCARD_CS_PIN = 4;

bool SD_Error = false;
#define dataFile "tstfile.txt"

File myFile;			// create instance of a "File" class

// *** Type K declarations ***************************************************
// ***************************************************************************

const uint8_t TK_CSPin = 2;
bool TypeK_Error = false;
SPISettings TypeKsettings(2000000, MSBFIRST, SPI_MODE0);

// *** setup() ***************************************************************
// ***************************************************************************

void setup() {

  pinMode (TK_CSPin, HIGH);
  digitalWrite(TK_CSPin,HIGH);
  pinMode (SDCARD_CS_PIN, HIGH);  // likely redundant, probably in SD.cpp
  digitalWrite(SDCARD_CS_PIN, HIGH);

  SerialUSB.begin(115200);
  delay (2000);

  // initialize SPI for Type K card
  SPI.begin();		// for MAX31855k aka TypeK

  // Initializing SD card....
  //pinMode (SDCARD_CS_PIN, OUTPUT);  // likely redundant, probably in SD.cpp
  if (!SD.begin(SDCARD_CS_PIN))
    {SerialUSB.print("initialization failed");
     SD_Error = true;
    }

  SerialUSB.print("READY TO OPEN FILE FOR WRITING = ");
  SerialUSB.println(dataFile);

  myFile = SD.open(dataFile, FILE_WRITE);
  // if the file opened okay, write to it:
  if (myFile) {
    myFile.println("data from boiler");
    myFile.close();
    }
  else {

    SD_Error = true;
    }
    SerialUSB.println((ReadRawData()>>2)*0.25);
}

void loop() {
  delay(2000);
  SerialUSB.println((ReadRawData()>>2)*0.25);

   myFile = SD.open(dataFile, FILE_WRITE);
  // if the file opened okay, write to it:
  if (myFile) {
    	myFile.println("data from boiler");
    	myFile.close();
    	}
  else {
    	SD_Error = true;
    	}
}

// *** Functions *********************************************************
// ***********************************************************************

uint16_t ReadRawData(void){
  digitalWrite(TK_CSPin, LOW);                                        //stop  measurement/conversion
  delayMicroseconds(5);
  digitalWrite(TK_CSPin, HIGH);                                       //start measurement/conversion
  delay(200);
  digitalWrite(TK_CSPin, LOW);                                        //set CS low to enable SPI interface for MAX31855
  uint16_t _rawData = 0;

  SPI.beginTransaction(SPISettings(TypeKsettings));
  _rawData = SPI.transfer16(0x0000); //read MSB
  SPI.endTransaction();
  digitalWrite(TK_CSPin, HIGH);
  return _rawData;
}  // --- ReadRawData ----
`

greiman, 2019-11-24:
SdFat calls SPI.beginTransaction() each time it accesses the SPI bus and SPI.endTransaction() to release the bus.  During this time SD cards have some non standard SPI behavior that requires extra clocks while SD chip select is high.  You device must not depend on SPI behavior while the SD card has the bus reserved in the interval between beginTransaction() and endTransaction().

Also do not set chip select low on a device before calling beginTransaction() and set chip select high before calling endTransaction().  beginTransaction() can cause a glitch on the bus when changing SPI settings.

JonRobert, 2019-11-24:
Thank you for the awesome fast response.  My board / devices are in pieces not due to another issue.  I will test as soon as I can.   I'm sending this so you know how much your response is appreciated :) :)
Will report back when I get this tested.   I will say dropping the CS until after the SPI.begin Transaction is non-intuitive.
John

closed 2019-11-24 by JonRobert

greiman, 2019-11-24:
Here is the reason you must not assert CS before transactionBegin().

This is the SAMD function that configures SPI when transactionBegin() is called.

```
void SPIClass::config(SPISettings settings)
{
  if (this->settings != settings) {
    this->settings = settings;
    _p_sercom->disableSPI();

    _p_sercom->initSPI(_padTx, _padRx, SPI_CHAR_SIZE_8_BITS, settings.bitOrder);
    _p_sercom->initSPIClock(settings.dataMode, settings.clockFreq);

    _p_sercom->enableSPI();
  }
}
```
I use a max SPI SCK rate for an SD and you use 2 Mhz so the test for a change in settings will cause SPI to be disabled while settings are changed then SPI is enabled.  This seems to cause a pulse on SCK.

Before you added the SD, the test for a change in settings prevented the pulse since there was no change in settings.