greiman/SdFat #170: 'myFile' does not name a type

mytja, 2019-12-20:
error: 'myFile' does not name a type
myFile.open("map1.h", O_READ);

greiman, 2019-12-21:
Don't have a clue from your limited info.  Have you declared myFile with something like this?

`  File myFile;`

mytja, 2019-12-22:
Yes, I did

greiman, 2019-12-23:
My crystal ball can't help. since you provided no code, no hardware info, and no IDE info.

Try this it should compile:

```
#include "SdFat.h"
File myFile;
void setup() {
  myFile.open("map1.h", O_READ);
}
void loop() {}
```
Output for an Uno.
```
Sketch uses 6264 bytes (19%) of program storage space. Maximum is 32256 bytes.
Global variables use 79 bytes (3%) of dynamic memory, leaving 1969 bytes 
for local variables. Maximum is 2048 bytes.
```
I am closing this since you provided no Info and I get about 20 requests for help every day with enough infor so I can help.

closed 2019-12-23 by greiman