greiman/SdFat #173: No Card, wrong chip select pin, or SPI problem?

liz7124, 2020-01-13:
I used Arduino Due R3 and Wiznet W5500 Ethernet Shield. I put micro SD Card 32 GB in the ethernet shield. When I ran my code and opened the serial monitor, it appears error like this:
`Can't access SD card. Do not reformat.`
`No card, wrong chip select pin, or SPI problem?`
`SD errorCode: 0X20,0XFF`
Did anyone get the same problem before? Any solutions?

greiman, 2020-01-13:
Ethernet shields often interfere with the SD unless the Ether Chip Select pin is high.

If the SD CS pin is 4 and the Ether CS pin is10, add the following code before initializing the SD.

```
  pinMode(10, OUTPUT);
  digitalWrite(10, HIGH);

  // Code to initialize the SD with CS pin 4 here.
```

xvinny, 2020-02-27:
Did you check the connector? Putting and removing the SD card several times damages the connector.