greiman/SdFat #174: Request for the void ls

MagoKimbra, 2020-01-26:
I apologize for the bad English and the request.
I use this library for the MK4duo fw (3D printers), unfortunately I am forced to import it into the FW and modify it only for the void ls, when I could use the downloadable one with arduino ide.
I use the printing of names with the file size, but the routine prints the size first and then the name. Hosts connected to the 3D printer want exactly the opposite, first the name and then the file size.

```
     if (flags & LS_DATE) {
        file.printModifyDateTime(pr);
        pr->write(' ');
      }
      if (flags & LS_SIZE) {
        file.printFileSize(pr);
        pr->write(' ');
      }
      file.printName(pr);
      if (file.isDir()) {
        pr->write('/');
      }
      pr->write('\r');
      pr->write('\n');
```

with

```
      file.printName(pr);
      if (file.isDir()) {
        pr->write('/');
      }
      if (flags & LS_DATE) {
        file.printModifyDateTime(pr);
        pr->write(' ');
      }
      if (flags & LS_SIZE) {
        file.printFileSize(pr);
        pr->write(' ');
      }
      pr->write('\r');
      pr->write('\n');
```
Thank's...

greiman, 2020-01-27:
I can't change ls() since it has worked this way for almost ten years.

I print name last since this is the way Windows dir and Linux ls work.  This insures date and size are column aligned.  Names can be up to 255 characters long.

You can simply provide a function to use place of ls() like this.
```
#include "SdFat.h"
const uint8_t chipSelect = SS;
SdFat sd;
bool myLs(FatFile *dir, print_t* pr, uint8_t flags, uint8_t indent = 0);
//------------------------------------------------------------------------------
bool myLs(FatFile *dir, print_t* pr, uint8_t flags, uint8_t indent) {
  FatFile file;
  if (!dir->isDir() || dir->getError()) {
    DBG_FAIL_MACRO;
    goto fail;
  }
  dir->rewind();
  while (file.openNext(dir, O_RDONLY)) {
    if (!file.isHidden() || (flags & LS_A)) {
    // indent for dir level
      for (uint8_t i = 0; i < indent; i++) {
        pr->write(' ');
      }
      file.printName(pr);
      if (file.isDir()) {
        pr->write('/');
      }      
      if (flags & LS_DATE) {
        file.printModifyDateTime(pr);
        pr->write(' ');
      }
      if (flags & LS_SIZE) {
        file.printFileSize(pr);
        pr->write(' ');
      }

      pr->write('\r');
      pr->write('\n');
      if ((flags & LS_R) && file.isDir()) {
        if (!file.ls(pr, flags, indent + 2)) {
          goto fail;
        }
      }
    }
    file.close();
  }
  if (dir->getError()) {
    goto fail;
  }
  return true;

 fail:
  return false;
}

void setup() {
  Serial.begin(9600);
  if (!sd.begin(chipSelect, SD_SCK_MHZ(50))) {
    sd.initErrorHalt();
  }
  // Use any directory here.
  File dir = sd.open("/");
  myLs(&dir, &Serial, LS_SIZE);
}
void loop() {
}
```

MagoKimbra, 2020-01-29:
ok, but that's what i already do ..
anyway the hosts of 3D printers, I tried some of them want first the name and then the size, otherwise in the file name they also put the value of the size.
Example with ls:
95456756 test.gco
in the host window I have the name 95456756 test.gco and the size empty.
Thanks anyway, I close ...

closed 2020-01-29 by MagoKimbra