greiman/SdFat #175: Versions >1.5.7 of the Adafruit SAMD core break SD initialization

prototypicalpro, 2020-01-28:
On an Adafruit Feather M0, using a a version of the Adafruit SAMD core greater than 1.5.7 cause the SD card to fail to initialize. I have also posted this issue in the [Adafruit SAMD core issue tracker](https://github.com/adafruit/ArduinoCore-samd/issues/209).

I tested this problem using v1.8.4 of the Arduino SAMD core and v1.1.1 of the SdFat library, uploading the `QuickStart` example sketch to an Adafruit Feather M0 WiFi with an Adalogger Featherwing and 16GB SDcard. No modifications were made to the sketch. Using version v1.5.7 of the Adafruit SAMD Core, I received the following output:
```
14:00:40.443 -> SPI pins:
14:00:40.443 -> MISO: 22
14:00:40.443 -> MOSI: 23
14:00:40.443 -> SCK:  24
14:00:40.443 -> SS:   16
14:00:40.443 -> 
14:00:40.443 -> Be sure to edit DISABLE_CHIP_SELECT if you have
14:00:40.443 -> a second SPI device.  For example, with the Ethernet
14:00:40.443 -> shield, DISABLE_CHIP_SELECT should be set to 10
14:00:40.443 -> to disable the Ethernet controller.
14:00:40.443 -> 
14:00:40.443 -> SD chip select is the key hardware option.
14:00:40.443 -> Common values are:
14:00:40.443 -> Arduino Ethernet shield, pin 4
14:00:40.443 -> Sparkfun SD shield, pin 8
14:00:40.477 -> Adafruit SD shields and modules, pin 10
14:00:40.477 -> 
14:00:40.477 -> Enter the chip select pin number: 10
14:00:48.490 -> 
14:00:48.490 -> Assuming the SD is the only SPI device.
14:00:48.490 -> Edit DISABLE_CHIP_SELECT to disable another device.
14:00:48.658 -> 
14:00:48.658 -> Card successfully initialized.
14:00:48.658 -> 
14:00:48.658 -> Card size: 15932 MB (MB = 1,000,000 bytes)
14:00:48.658 -> 
14:00:48.658 -> Volume is FAT32, Cluster size (bytes): 32768
14:00:48.658 -> 
14:00:48.658 -> Files found (date time size name):
14:00:48.658 -> 2000-01-01 01:00:00      71899 HYDRO11.CSV
14:00:48.658 -> 2000-01-01 01:00:00       6265 test00.csv
14:00:48.658 -> 2019-09-05 15:51:34        587 Google_H.txt
14:00:48.658 -> 2019-09-05 15:51:34        583 Google_O.txt
14:00:48.658 -> 2019-09-05 15:51:34        594 Google.txt
14:00:48.658 -> 2019-09-05 15:51:34        450 Max.txt
14:00:48.692 -> 2000-01-01 01:00:00      10579 test01.csv
14:00:48.692 -> 2000-01-01 01:00:00       9962 test02.csv
14:00:48.692 -> 2000-01-01 01:00:00      44594 test03.csv
14:00:48.692 -> 
14:00:48.692 -> Success!  Type any character to restart.
```
However when upgrading to version v1.5.9, the example no longer worked:
```
13:49:43.446 -> SPI pins:
13:49:43.446 -> MISO: 22
13:49:43.446 -> MOSI: 23
13:49:43.446 -> SCK:  24
13:49:43.446 -> SS:   16
13:49:43.446 -> 
13:49:43.446 -> Be sure to edit DISABLE_CHIP_SELECT if you have
13:49:43.446 -> a second SPI device.  For example, with the Ethernet
13:49:43.446 -> shield, DISABLE_CHIP_SELECT should be set to 10
13:49:43.480 -> to disable the Ethernet controller.
13:49:43.480 -> 
13:49:43.480 -> SD chip select is the key hardware option.
13:49:43.480 -> Common values are:
13:49:43.480 -> Arduino Ethernet shield, pin 4
13:49:43.480 -> Sparkfun SD shield, pin 8
13:49:43.480 -> Adafruit SD shields and modules, pin 10
13:49:43.480 -> 
13:49:43.480 -> Enter the chip select pin number: 10
13:49:45.784 -> 
13:49:45.784 -> Assuming the SD is the only SPI device.
13:49:45.784 -> Edit DISABLE_CHIP_SELECT to disable another device.
13:49:45.819 -> 
13:49:45.819 -> SD initialization failed.
[... 258 lines cut ...]
Generating function prototypes...
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"QuickStart.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0 -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0\"" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -DARDUINO_SAMD_ZERO -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0 -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0\"" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\variants\\feather_m0" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\libraries\\SPI" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\libraries\\Adafruit_ZeroDMA" "-IC:\\Users\\User\\Documents\\Arduino\\libraries\\SdFat\\src" "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_15175\\sketch\\QuickStart.ino.cpp" -o "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_15175\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_15175\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD "-D__SKETCH_NAME__=\"\"\"QuickStart.ino\"\"\"" -DF_CPU=48000000L -DARDUINO=10810 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0 -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0\"" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -DARDUINO_SAMD_ZERO -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0 -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0\"" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\variants\\feather_m0" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\libraries\\S [... cut ...] \\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -DARDUINO_SAMD_ZERO -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0 -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0\"" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.9\\cores\\arduino/TinyUSB" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.5.

greiman, 2020-01-29:
I have wasted too much time on Adafruit SAMD SPI bugs.

Try the [Adafruit fork of Sdfat](https://github.com/adafruit/SdFat)..

prototypicalpro, 2020-01-29:
I tried using 12MHz or 1MHz speeds, and using the Adafruit SdFat fork, but still had the same issue.

Sorry to hear that. I can close this issue in favor of the one on Adafruit SAMD core?

bkw777, 2020-09-15:
THANK YOU for ending my hair-pulling. I just hit this too.

I have pretty simple code that has been working for a couple years, that I just recently started working on again, and stopped working, but only on Feather M0 Adalogger. Feather 32u4 Adalogger and Teensy 3.5 & 3.6 all kept working. (teensy is using SDIO though), so of course I assumed I broke it.

At least now I know to stop trying to find what's wrong with my code and I can just document the problem and work-around and move on for now.

bkw777, 2020-09-15:
This worked for me:
<pre>
...
#elif defined(ADAFRUIT_FEATHER_M0)
...
#define SD_SPI_MHZ 12   // https://github.com/adafruit/ArduinoCore-samd/pull/186
...
 #if defined(SD_CS_PIN) && defined(SD_SPI_MHZ)
    if (SD.begin(SD_CS_PIN,SD_SCK_MHZ(SD_SPI_MHZ))) {
 ...
</pre>