greiman/SdFat #177: Extremely slow SD write speeds on STM32?

Bambofy, 2020-03-05:
Hi,

I am using this library on an STM32 L452RE however the write speeds i'm receiving are very slow, 8KB/s. See this thread: https://www.stm32duino.com/viewtopic.php?f=7&p=1409

greiman, 2020-03-05:
The STM32duino SPI library is very slow for single byte transfers.  A custom SPI driver that uses multi-byte DMA transfers would help.  See examples in SdFat/src/SpiDriver.

You will never achieve a reliable write latency short enough with a single thread to read samples and write to the SD.  SD cards can occasionally have a write latency of about 100 ms.

You will need to read samples in a timer callback routine and queue the data to the write loop.

This [example for an Uno](https://github.com/greiman/SdFat/tree/master/examples/AnalogBinLogger) can sample a 10-bit ADC at 16,000 HZ.  The rate is a bit faster since the sample interval is 62 usec.

Here is output from the example:

```
Sample pin: 0
ADC bits: 10
ADC clock kHz: 500
Sample Rate: 16129.03
Sample interval usec: 62.0000
Creating new file
Erasing all data
Logging - type any character to stop
Truncating file
File renamed: analog01.bin
Max block write usec: 928
Record time sec: 27.508
Sample count: 443660
Samples/sec: 16128.40
Overruns: 0
Done
```

Bambofy, 2020-03-05:
ah ok thank you for the tips, i will look into the DMA transfer systems for STM32 and see if i can figure it out :)

With respect to the write buffering, how does this work considering the SD writing would be a lot slower than the ADC sampling? Wont the ADC sampling wrap around and overrun the SD writing, causing a delay anyway?

greiman, 2020-03-05:
Look at the [example in the above link](https://github.com/greiman/SdFat/tree/master/examples/AnalogBinLogger).  It has a buffer queue.  Data is placed in the queue in the interrupt routine and written to the SD in loop.  The [second example](https://github.com/greiman/SdFat-beta/tree/master/examples/AvrAdcLogger) is even better. 

Read the output from the example above.  I ran it for 27 seconds but have tested it for many minutes.  It took 443,660 samples in 27.5 second.  This was on an Uno.

Bambofy, 2020-03-05:
Amazing! I will try this technique

Bambofy, 2020-03-06:
Got it working using the FIFO technique :) I can sample an ADC at 16,000Hz at 16bit depth for >10minutes, since i'm using the STM32 L452RE i can use a FIFO buffer size of 81960 bytes!

Bambofy, 2020-03-16:
@greiman Hi, i've found that using this way https://github.com/greiman/SdFat/blob/master/examples/AnalogBinLogger/AnalogBinLogger.ino sometimes corrupts the SD card volume so that .createContiguous fails.

Do you know how to make sure that [image]  doesn't overwrite any critical file data?

greiman, 2020-03-16:
I have no idea what what you are doing.  I can't help debug your code.

Bambofy, 2020-03-16:
> 
> 
> I have no idea what what you are doing. I can't help debug your code.

Ok no problem, i'm trying to debug it myself but i can't find the source of the bug.