greiman/SdFat #180: Incompatible with MiniCore SPI1.h

vpreatoni, 2020-03-26:
SdFat seems to be incompatible with MiniCore ATMEGA328PB hardware.

MiniCore defines SPI.h if you need to use first SPI port, and SPI1.h if 2nd port needed.
If using 2nd port, SPI class is defined as SPIClass1 * SPI, so needs to be casted as follows:

`SdFat sdCard((SPIClass*)&SPI1);`

But, at some point, after executing sdCard.begin(SS1, SD_SCK_MHZ(8)); SdFat lib tries to initialize 1st SPI port too, since I lost GPIOs of SPI (SCK, MISO, MOSI). SdFat lib should not touch those pins, as I'm instructing lib to use SPI1 (SCK1, MISO1, MOSI1).

Any workaround for this?? STM32 example is not valid for 328PB:
```
// Use second SPI port
SPIClass SPI_2(2);
```
SPIClass  constructor on 328PB does not accept any parameters.

greiman, 2020-03-26:
Looks like support of SPI1 is not possible.  The board package defines two SPI classes.  The standard way is to define one SPI class with several instances of the class.  See the SDMD board packages by Arduino.

If there was a single SPI class you could edit SdFatConfig.h at about line 72 to enable multiple SPI ports.

I suspect a cast won't work.

```
/**
 * If the symbol USE_STANDARD_SPI_LIBRARY is zero, an optimized custom SPI
 * driver is used if it exists.  If the symbol USE_STANDARD_SPI_LIBRARY is
 * one, the standard Arduino SPI.h library is used with SPI. If the symbol
 * USE_STANDARD_SPI_LIBRARY is two, the SPI port can be selected with the
 * constructors SdFat(SPIClass* spiPort) and SdFatEX(SPIClass* spiPort).
 */
#define USE_STANDARD_SPI_LIBRARY 0  // <<-- set to two for  multiple ports.
```
I will not support multiple SPI classes.

SdFat-beta can support user SPI drivers but it is not well tested.  [Here is an example](https://github.com/greiman/SdFat-beta/tree/master/examples/UserSPIDriver).

You could try replacing all occurrences of SPI with SPI1 in the example driver.

vpreatoni, 2020-03-26:
Tnx for your answer!. I've reported this to MiniCore devs, as having a different class for each SPI port doesn't seem right to me neither. [https://github.com/MCUdude/MiniCore/issues/132](url)

Will give a try to SdFat-beta and report any issues on that repo.