greiman/SdFat #181: SdFat conflict with SPIFFS

williamesp2015, 2020-04-08:
I am using SPIFFS to read and write SPIFFS on ESP32 using Arduino platform. I have seriouse problem with File definitions;
File logFile = SPIFFS.open(FILENAME_DATALOG, FILE_WRITE);
LowLatencyLogger.cpp:658:56: error: invalid conversion from 'int' to 'const char*' [-fpermissive]

greiman, 2020-04-08:
I have no idea since I don't use esp32.

uzi18, 2020-04-08:
@williamesp2015 please paste rest of error

williamesp2015, 2020-04-09:
Hi @uzi18. If FS.h change class File name it will solve the problem. I tried to change in my project but other files like WebServer.cpp uses FS.h too.
Complete errors:
src\FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'
C:/users/am/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:47:7: error: previous definition of 'class fs::File'
src\FatLib/FatFileSystem.h:95:13: error: 'class fs::File' has no member named 'open'

nodoubtman, 2022-01-30:
> 

What are the exact definitions ?
I have the same problem.

Thanks.
Marc.

greiman, 2022-01-30:
File is only defined here in SdFat.h at about line 450
```
//
// Only define File if FS.h is not included.
// Line with test for __has_include must not have operators or parentheses.
#if defined __has_include
#if __has_include(<FS.h>)
#define HAS_INCLUDE_FS_H
#warning File not defined because __has_include(FS.h)
#endif  // __has_include(<FS.h>)
#endif  // defined __has_include
#ifndef HAS_INCLUDE_FS_H
#if SDFAT_FILE_TYPE == 1 || defined(DOXYGEN)
/** Select type for File. */
typedef File32 File;
#elif SDFAT_FILE_TYPE == 2
typedef ExFile File;
#elif SDFAT_FILE_TYPE == 3
typedef FsFile File;
#endif  // SDFAT_FILE_TYPE
#endif  // HAS_INCLUDE_FS_H
```
You can just use File32, ExFile, or FsFile and never define File by disabling or removing this section.

File was added to the Arduino port of SdFat in 2009 so I added a typedef to be backward compatible.

nodoubtman, 2022-01-30:
Thanks for answering.
I dont understand.
What i have to do to use spiffs with sdfat ?
Thanks.
Have a good day.
Marc.

greiman, 2022-01-30:
> What i have to do to use spiffs with sdfat ?

The simplest thing is to do what @williamesp2015 did.

Change File to File_ in the above section of SdFat.h and use File_ in your app.

You can also delete the above section of SdFat.h and use the appropriate class name, File32, ExFile, or FsFile.

ockernuts, 2023-03-10:
If fs::FS compatibility is needed, this can be acchieved by writing an adapter: see https://github.com/greiman/SdFat/issues/148#issuecomment-1464448806