greiman/SdFat #182: 508 bytes of data in 512 byte block size - Purpose?

censix, 2020-04-27:
Hi Bill

I have been using SdFat for several months now. I acquire ADC data at about 40k samples per second and write directly to an attached SD card. Works great!
... just one thing.
I acquire 254 samples(byte) in one data record. So two records fit into the  508 = 2*254
bytes for one block of 512, to be written to the SD card, using your library.

https://github.com/greiman/SdFat/blob/master/examples/AnalogBinLogger/AnalogBinLogger.h
// Data block for 8-bit ADC mode.
const size_t DATA_DIM8 = 508;

After looking closely at the binary log file on the card, I see that the last 4 bytes (509-512) of each block, apparently written by the SDFat library, never change. Their values are: 05 AF 00 0D.
Since all blocks end with these same four bytes, no matter what the data content of the block is, I am wondering why SDFat needs them.  They obviously don't contain a checksum of the data bytes, so why have them? I am asking since I would really like to use these four bytes for writing data values as well!

Q: Is there any specific reason why I should NOT use the last 4 bytes of each block for data? Will SDcard writing/reading fail if if do write data to them?

Thanks!

greiman, 2020-04-27:
Entire 512 byte SD blocks must be written so the size the of struct block8_t or block16_t is 512 bytes.  

Depending on the pin count, part of the block may not be used. The first four bytes are used as an overrun count and a count of ADC values in the block.

censix, 2020-04-27:
"Depending on the pin count, part of the block may not be used"
I see. Well I am only reading from one pin. So this should not be an issue.
I am now simply processing the .bin files with another step that cuts off the first 4 and the last 4 bytes. The result is acceptable for me.
Cheers

closed 2020-04-27 by censix