greiman/SdFat #183: Define SPI for use

michaliskon, 2020-04-28:
I have looked into the STM32Test example and I am trying to define an SPI bus to use in ESP32 w Arduino IDE.

`  
    #define IMPLEMENT_SPI_PORT_SELECTION 1
SPIClass SD_SPI(HSPI);
  SdFat sd(&SD_SPI);
`

Compiling gives 

`
Home_server:146:21: error: no matching function for call to 'SdFat::SdFat(SPIClass*)'

     SdFat sd(&SD_SPI);

                     ^

In file included from ...\coding\Arduino IDE\Home_server\Home_server.ino:34:0:

...\coding\Arduino IDE\libraries\ESP32_SdFat\src/SdFat.h:310:7: note: candidate: SdFat::SdFat()

 class SdFat : public SdFileSystem<SdSpiCard> {

       ^

...\coding\Arduino IDE\libraries\ESP32_SdFat\src/SdFat.h:310:7: note:   candidate expects 0 arguments, 1 provided

...\coding\Arduino IDE\libraries\ESP32_SdFat\src/SdFat.h:310:7: note: candidate: constexpr SdFat::SdFat(const SdFat&)

...\coding\Arduino IDE\libraries\ESP32_SdFat\src/SdFat.h:310:7: note:   no known conversion for argument 1 from 'SPIClass*' to 'const SdFat&'

...\coding\Arduino IDE\libraries\ESP32_SdFat\src/SdFat.h:310:7: note: candidate: constexpr SdFat::SdFat(SdFat&&)

...\coding\Arduino IDE\libraries\ESP32_SdFat\src/SdFat.h:310:7: note:   no known conversion for argument 1 from 'SPIClass*' to 'SdFat&&'
`

greiman, 2020-04-28:
I have not implemented SPI port selection for ESP32.

You could try the following.

At line 80 of SdFatConfig.h set USE_STANDARD_SPI_LIBRARY to 2 like this.

`#define USE_STANDARD_SPI_LIBRARY  2  // 0`

This at least causes the following to compile.

```
#include "SdFat.h"
SPIClass SD_SPI(HSPI);
SdFat sd(&SD_SPI);

void setup() {
  sd.begin();
}

void loop() {
}
```
If this fails, I don't have time to this problem.

michaliskon, 2020-04-29:
Works like a charm. Thank you greiman

marcosQuesada, 2020-09-12:
I'm facing the same issue on a STM32F103C8, i just tried the config change (#define USE_STANDARD_SPI_LIBRARY 2 // 0), but i''m still getting "no matching function for call to 'SdFat::SdFat(SPIClass (*)())'"
Any hint will be much appreciated :)

iharshadec, 2021-02-27:
@greiman 
I do not see this line 80. Is this changed in latest release? I am not able to begin SD using ESP32 with HSPI pin connections. It works fine if I choose VSPI but not when select HSPI.

greiman, 2021-02-27:
I added an option to the ESP driver to use HSPI in V2 like this:

```
SPIClass SD_SPI(HSPI);
#define SPI_CLOCK SD_SCK_MHZ(50)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK, &SD_SPI)

...

  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
```
It seems unstable.  You can try a slower clock or try the library driver by editing SdFatConfig.h at about line 115.

#define SPI_DRIVER_SELECT 1 // 0 change from 0 to 1.

iharshadec, 2021-02-28:
Thank you @greiman for this notes. 
I was using SdFat-beta and now as this is merged to this main library I was trying to get this working. Yes I have observed that for ESP32 even with VSPI it's not working with 50Mhz it is stable only with 25Mhz.
I will give this a try and see how it goes and will update here.