greiman/SdFat #185: Issue with .createContiguous

Bambofy, 2020-05-06:
See the original issue here: https://github.com/greiman/SdFs/issues/7

Do you know what would happen if the Arduino were to power off between writeStart and writeEnd?

greiman, 2020-05-06:
The result would be undefined.  Each SD card has some RAM and data would be lost.

Bambofy, 2020-05-07:
I've found that  if i run the program multiple times, the SD card fixes itself.

Start 1:
1. sd card ok
Start 2:
1. sd card cannot be used
Start 3:
1. sd card ok

Does this mean anything to you? i'm not sure why it would fix itself over restarts?

Finally, what would be the correct way to handle .createContiguous returning false? Should i erase and format the entire card?

greiman, 2020-05-07:
Always reformat a card if you can't read or write files.

Bambofy, 2020-05-07:
> Always reformat a card if you can't read or write files.

Thanks :)

closed 2020-05-07 by Bambofy