greiman/SdFat #189: Does not compile with new STM32 Core package from ST

ShaggyDog18, 2020-05-16:
get errors compiling with the new STM32 core framework from ST: [](https://github.com/stm32duino/Arduino_Core_STM32)
Error for the included STM32Test example using **Generic STM32F1 series**:
`STM32Test:17:17: error: no matching function for call to 'SPIClass::SPIClass(int)'
   17 | SPIClass SPI_2(2);
      |                 ^
In file included from C:\Users\xyz\Documents\Arduino\Libraries\SdFat\examples\STM32Test\STM32Test.ino:5:
`
Error for my project:
> undefined reference to `SdSpiCard::begin(SdSpiAltDriver*, unsigned char, SPISettings)'

greiman, 2020-05-16:
The STM32Test example does not support the STM32 Core framework.  It is for the Roger Clark STM32 package.

I don't plan to offer a custom SPI driver for Arduino_Core_STM32.  There is no way I can support custom drivers in the future for all Arduino board packages.

You might be able to define a second SPI port using the Arduino_Core_STM32 SPI library like this:

```
//      define for your board      MOSI1  MISO1  SCK1
SPIClass SPI1(MOSI 1, MISO 1, SCK1);
```

Then edit SdFatConfig.h and set USE_STANDARD_SPI_LIBRARY like this for port selection.

`#define USE_STANDARD_SPI_LIBRARY 2`

I have not tested this.

ShaggyDog18, 2020-05-17:
I do not need a second SPI for my project, just risen your attention to the issue.
As for my project: I use STM32F1 Blue Pill and one standard SPI1; keep getting the following error:
`c:/users/xyz/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe:` sketch\mySketch-v5i.ino.cpp.o: in function `setup':
mySketch-v5i.ino.cpp:(.text.setup+0x84): undefined reference to `SdSpiCard::begin(SdSpiAltDriver*, unsigned char, SPISettings)'
`

greiman, 2020-05-17:
Can't help from that message.

devel0, 2021-05-16:
This worked for me on stm32f446re arduino platformio,

```cpp
static SPIClass mySPI2(PB15, PB14, PB13, PB12); // set for SPI2
SdFs sd;
sd.begin((SdSpiConfig(PB12, DEDICATED_SPI, SD_SCK_MHZ(18) , &mySPI2)));
```