greiman/SdFat #192: SdFile::dateTimeCallback(dateTime); Hour is not correct

Primus007, 2020-05-25:
For a test, i have set the hour, minute and second to 14
for testing i have 7 different files opened for write. and closed .
`      hour = 14;
  minute = 14;
  second = 14;SdFile::dateTimeCallback(dateTime);
    uploadFile.open((char *)filename.c_str(), FILE_WRITE);     `

3 of the 7 files has the wrong hour. it is +1 or sometimes -1 hour.
[image]
in this test the wrong  hour is always -1 hour.

is this a bug?

greiman, 2020-05-25:
I have no Idea what you are doing.

closed 2020-05-25 by greiman

Primus007, 2020-05-25:
I set the File Timestamp with SdFile::dateTimeCallback(dateTime) 
This is a function in sdfile.h see from line 264 in sdfile.h "date/time callback function"
But the hour is sometimes not correct.

greiman, 2020-05-26:
Don't call timestamp.  Setting the callback is all you need to do.

Primus007, 2020-05-26:
Without this line
uploadFile.timestamp(T_CREATE, year, month, day, hour, minute, second);`

the problem is the same.
All works fine ONLY the hour is sometimes +1 or -1 . see screenshot in my first post.
this must be a bug. please can you check this?

greiman, 2020-05-27:
I have no idea what you screenshot is.  The one hour difference appears to be associated with the file day month year.

FAT has no concept of daylight/summer time.  So programs like Linux sometimes try to adjust by an hour.  This won't work unless you adjust the hour in the callback by +1, -1.

Try using SdFat ls() and see if the hour is what you set.

Thousands of users timestamp their files with a RTC call back and it works.

Don't bother me, I can't help you!