greiman/SdFat #195: sdfat error code for sdinfo

ryuk171, 2020-06-22:
i have connected al pins appropriately for nodemcu8266 v2.0 cp2102 , still sdcard error
pins
cs - d8
sck - d5
mosi - d6
miso - d7
vcc - 5v
gnd - gnd
 serial monitor log
13:48:07.731 ->  ⸮/⸮⸮4s⸮⸮
13:48:08.002 -> 
13:48:08.002 -> SdFat version: 10102
13:48:08.002 -> 
13:48:08.002 -> Assuming the SD is the only SPI device.
13:48:08.071 -> Edit DISABLE_CHIP_SELECT to disable another device.
13:48:08.104 -> 
13:48:08.104 -> Assuming the SD chip select pin is: 15
13:48:08.137 -> Edit SD_CHIP_SELECT to change the SD chip select pin.
13:48:08.206 -> 
13:48:08.206 -> type any character to start
13:48:12.736 -> error: cardSize failed
13:48:12.736 -> SD errorCode: 0X50,0X20
13:48:12.770 -> 

i tried the readwrite example it said initialization failed

greiman, 2020-06-22:
Can't help since it is a hardware read error and I don't know anything about your board.  

You could try a lower SCK rate.

closed 2020-06-22 by greiman

ryuk171, 2020-06-22:
it works fine with the normal sd(esp8266) example , here only i see this error log 

[https://photos.google.com/share/AF1QipOY2drGHkI54xBcD1XfuLNEEUYwaPSBngA-zB00kxsEWyJ2fBuuXPn36usi47GVhA?key=dUNJMjFpRzR0Wmc1RlNlU3dUWGlTQ0I2UndIcWdn](url)

the sdcard module

greiman, 2020-06-22:
Use the "normal(esp8266) library"  It is an older version of SdFat ported by Earle F. Philhower.

https://github.com/earlephilhower/ESP8266SdFat