greiman/SdFat #198: I can initialize the card and list files but I cannot create or open a file

afreisinger, 2020-07-26:
I can initialize the card and list files but I cannot create or open a file.
I use Arduino MEGA and W5100 with ReadWrite.ino example
I testing QuickStart.ino too and it work well.
 I took the the precaution of putting pin 10 as the output and high and pin 53 as output too.

`#define SD_CS_PIN 4`
`#define DISABLE_CHIP_SELECT 10`
`#define SS 53`
`pinMode(DISABLE_CHIP_SELECT, OUTPUT);`
`digitalWrite(DISABLE_CHIP_SELECT, HIGH);`
`pinMode(SD_CS_PIN, OUTPUT);`
`digitalWrite(SD_CS_PIN, LOW);`
`pinMode(SS, OUTPUT);`

but the output not change

`Initializing SD card..initialization done.`
`error opening test.txt`
`error opening test.txt`

greiman, 2020-07-26:
Looks like you have the correct pin modes for pin 10 and 53.

It might be the SPI frequency.  The W5100 supports a max SPI frequency of 4 MHz.

Try using this code to initialize the card:

```
#define SPI_SPEED SD_SCK_MHZ(4)

...

  if (!sd.begin(SD_CS_PIN , SPI_SPEED)) {
    // handle error 
  }
```

afreisinger, 2020-07-26:
Nothing change, I try with 4 and 2 Mhz.
I can initialize the card without problems but can't open file.

I am disoriented. Will the sd card slot be broken?

SPI pins:
MISO: 50
MOSI: 51
SCK:  52
SS:   53

SD chip select is the key hardware option.
Common values are:
Arduino Ethernet shield, pin 4
Sparkfun SD shield, pin 8
Adafruit SD shields and modules, pin 10

Enter the chip select pin number: 4
Disabling SPI device on pin 10

Card successfully initialized.
Card size: 1978 MB (MB = 1,000,000 bytes)
Volume is FAT16, Cluster size (bytes): 32768
Files found (date time size name):
2020-07-25 22:35:42          0 Data00.csv

Success!  Type any character to restart.

greiman, 2020-07-27:
I can't help more since I don't have your hardware.

QuickStart opens files to list them.

GunZi200, 2020-07-27:
I am also using ATmega2560 but with W5500. Does `digitalWrite(53, HIGH)` make any difference? I've haven't run into any trouble so far.

Also try moving  `pinMode(SS, OUTPUT);` above `digitalWrite(SD_CS_PIN, LOW);`

The default SPI frequency for W5100 when using the Ethernet library is 14MHz so I don't think thats your problem. If the above doesn't work I would suggest doing some measurements on the SPI lines.

What hardware are you using with W5100?

greiman, 2020-07-27:
The Arduino W5500 shield and clones are reliable.  The original Arduino W5100 shield had resistor voltage dividers to convert 5V to 3.3V for SCK, MOSI, and chip select.  These shields only worked well at lower SCK rates.

You should be able to open files if QuickStart can list files with your shield.

afreisinger, 2020-07-30:
I concluded that it is a hardware problem, I don't know, possibly the sd card socket. I connected another SD module directly to pin 4 of the ethernet module and was able to initialize, open and write to the file. I don't know why, with the previous scheme it allowed me to initialize the card. Thank you very much.

closed 2020-07-30 by afreisinger