greiman/SdFat #206: 24Mhz on SAMD arduino Zero

gioreva, 2020-09-10:
Changing the line 45
https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SPI/SPI.h
on
#define SPI_MIN_CLOCK_DIVIDER (uint8_t)(1 + ((F_CPU - 1) / 24000000))

And calling sd.begin(12, SD_SCK_MHZ(24) );

SD stop working

Calling sd.begin(12, SD_SCK_MHZ(12) )  Work well.

Does anyone have experience if there are problems with the SdFat library or Arduino SPI library ?

greiman, 2020-09-10:
SAMD just can't reliably support 24 MHz.  That's why the clock divider was limited to 12 MHz in the SPI library.

gioreva, 2020-09-10:
Changing the line 45 of the SPI Library
https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SPI/SPI.h
on
#define SPI_MIN_CLOCK_DIVIDER (uint8_t)(1 + ((F_CPU - 1) / 24000000))

SPI is 24Mhz.
Signal, with oscilloscope are good.

https://snipboard.io/7f5uxT.jpg

greiman, 2020-09-10:
Not my problem, talk to the author of the SPI library.  Some SAMD devices did work at 24 MHz but many failed.  I won't modify SdFat since the official Arduino library supports a max of 12 MHz.

[See this.](https://github.com/arduino/ArduinoCore-samd/pull/292#issuecomment-361976681) 

> Moreover, at the times, an Atmel' engineer confirmed that the maximum clock speed for the SAMD21 is 12MHz.

bkw777, 2020-09-15:
This worked for me. 2 different boards.
<pre>
...
#elif defined(ADAFRUIT_FEATHER_M0)
...
#define SD_SPI_MHZ 12   // https://github.com/adafruit/ArduinoCore-samd/pull/186
...
 #if defined(SD_CS_PIN) && defined(SD_SPI_MHZ)
    if (SD.begin(SD_CS_PIN,SD_SCK_MHZ(SD_SPI_MHZ))) {
...
</pre>