greiman/SdFat #207: slow freeClusterCount

geologic, 2020-09-22:
Hi

I'm using SdFat with arduino@8Mhz and a SPI microSD card reader.
At startup i check for a SD card and print free space with freeClusterCount(). With a 4Gb class 10 SD empty card it takes around 18 seconds to get free space, is any way to speed up the free space reading?
There is an option MAINTAIN_FREE_CLUSTER_COUNT but it seems it only improves speed after the first call to freeClusterCount(), that's right? In my case i only call freeClusterCount() once...

greiman, 2020-09-22:
The first count must read the entire FAT.  A 4GB card requires reading 1000 blocks of 512 byte each and scanning each block.

geologic, 2020-09-23:
I saw your SdFat-Beta library, do you think it can be faster to calculate free space?
I only need some basic functions like check free space and write a file to SD.

greiman, 2020-09-23:
I suspect there will be little difference.

geologic, 2020-09-23:
ok, thanks.

closed 2020-09-23 by geologic