greiman/SdFat #210: A few questions about "minimum necessary actions" for performing raw write

jerabaul29, 2020-10-07:
Many thanks for a very nice library.

I want to perform some raw write at high speed / low latency and I have a few questions.

My starting point is your example script:

https://github.com/greiman/SdFat/blob/master/examples/RawWrite/RawWrite.ino

Can you confirm that I I understand well that this program:

- creates a contiguous file of pre-determined size (guess this improves performance rather than extending "on the fly"?

https://github.com/greiman/SdFat/blob/7d0261d144f4c18d81884c1fd0c8c78e1567fa80/examples/RawWrite/RawWrite.ino#L77-L79

- makes the card ready to write blocks one after the other:

https://github.com/greiman/SdFat/blob/7d0261d144f4c18d81884c1fd0c8c78e1567fa80/examples/RawWrite/RawWrite.ino#L105-L108

- writes blocks 1 at a time (it seems that quite a few things happen in the background here, like automatically writing to the next block each time this is called, right?)

https://github.com/greiman/SdFat/blob/7d0261d144f4c18d81884c1fd0c8c78e1567fa80/examples/RawWrite/RawWrite.ino#L149-L151

I have a few questions:

- it seems that the library has an internal buffer of size the block size ie 512 bytes that is used for writing:

https://github.com/greiman/SdFat/blob/7d0261d144f4c18d81884c1fd0c8c78e1567fa80/examples/RawWrite/RawWrite.ino#L90

https://github.com/greiman/SdFat/blob/7d0261d144f4c18d81884c1fd0c8c78e1567fa80/examples/RawWrite/RawWrite.ino#L149

I suppose that this buffer is not created if not the vol() command is used, is that right? Also, Would it be possible to call the writeData function on a custom buffer? Will it well just write the next 512 bytes on the buffer to the SD card?

jerabaul29, 2020-10-07:
Regarding the last question: in other words; if  I have a buffer of bytes

```
unsigned char buff[1024];
```

Will:

``` 
sd.card()->writeData(buff)
```

write the first 512 bytes of the buffer and

```
sd.card()->writeData(buff+512)
```
write the following 512 bytes?

greiman, 2020-10-07:
The rawWrite  example was written many years ago.  Modern cards have very large flash pages, up to  512KB, and only emulate 512 byte pages.  This example does not produce high performance and low latency with these cards.

The internal 512 byte buffer is required to open/create files so is always created.

For high performance it is best to use SdFat-beta with dedicated SPI.  This takes advantage of the large flash pages.

See the [ExFatLogger](https://github.com/greiman/SdFat-beta/blob/master/examples/ExFatLogger/ExFatLogger.ino) example with dedicated SPI.

jerabaul29, 2020-10-07:
Thanks! So maybe I should first try with this library as it is not in beta any longer so I expect it to be more stable, and if it is not good enough, move to the beta one?

greiman, 2020-10-07:
I will close this - your on your own to get performance since the new SdFat is a total rewrite.

closed 2020-10-07 by greiman